prusa3d/Prusa-Firmware #2892: [BUG] [FW 3.x.x] the belt test brings the value 0

Blumenkroete, 2020-11-15:
**Printer type** - [MK3s]
**Printer firmware version**-  [3.9.2]

**MMU Upgrade** - [MMU2S]
**MMU upgrade firmware version [1.0.6-372]

**SD card or USB/Octoprint**
Doesn't matter

**Describe the bug**

(first of all = the printer works **without** problems )
(i get around 30 days on workload on it )

My "Belt Test" always gives me 0 values

No matter which FW I use 

Selftest, Reset to factory settings, With and Without the MMU Unit 
the Belt-status are still 0 

So the XYZ calibration did not bring a fix 
  
**To Reproduce**
Run the Selftest > pass > Belt status 0 
  
**Expected behavior**
to get an value for the belt 

**G-code**
Doesn't matter

**Video**
https://youtu.be/-It22xHuJ8o

leptun, 2020-11-15:
I think this might be a hardware issue. I’ve stumbled across something similar recently. Please try to do the following to see if it’s the same:
Flash an official prusa firmware using prusaSlicer (very important) and then try to switch to another language. Does it switch or is it always english?

Blumenkroete, 2020-11-15:
i cannot change the language because i have the "Einsy Rambo rev.10.a 
Therefore I have the "External SPI flash W25X20CL not responding - error". 

https://help.prusa3d.com/en/article/external-spi-flash-w25x20cl-not-responding-error_112019

susajr, 2020-11-16:
Hello, 
please contact us on email info@prusa3d.com, write "For Honza" into the subject and send a photo of the EINSY so we can see the revision number + your printer order number, we will try to help.

Blumenkroete, 2020-11-16:
@susajr ok thanks

JohnnyDeer, 2021-07-19:
Hi @Blumenkroete do you have any update on this issue? Does our tech support helped you resolve it?

Blumenkroete, 2021-07-31:
Hey! 

no the bug is still there but the Printer work very well

closed 2023-11-02 by github-actions[bot]