prusa3d/Prusa-Firmware #2894: [BUG] 3.9.0, 3.9.1 - issue with fine details printing

eliotik, 2020-11-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0, 3.9.1

**MMU Upgrade** - no
**MMU upgrade firmware version - no

**SD card or USB/Octoprint**
Bug present in both variants, when printing from SD or with octoprint

**Describe the bug**
I created a post on reddit here: https://www.reddit.com/r/prusa3d/comments/hkriyf/help_needed_nozzle_pushes_on_print_while/
Listed all details of debugging  the issue, also w/ and w/o support help.
Issue is that with 3.9.* when I try to print cones, or any other models with fine details - printer pushes with nozzle on these details.
When i downgrade back to 3.8.1 - everything is fine.
This comment is a result of my conversation with support: https://www.reddit.com/r/prusa3d/comments/hkriyf/help_needed_nozzle_pushes_on_print_while/fxlg0ru/?utm_source=reddit&utm_medium=web2x&context=3
Copying it here:
```
Today I was talking with support about another problem and mentioned this issue and I was asked if I did factory reset after upgrade which I didn't remember.

So I created a small model with cones, printed it while on 3.8.1. Upgraded to 3.9, did factory reset, calibrated everything. Printed same gcode file as on 3.8.1.

And result was again the same as in the past - cones are smashed:

https://imgur.com/SYu0Ouh

on top is result of 3.9

on bottom is result of 3.8.1

So it's not filament or nozzle sock or ambient temperature.

3mf and gcode files can be found here: https://gofile.io/d/raG0Wl

Was on a chat with support, tried to print gcode for the same model but Georgios modified it and sent gcode to me: https://gofile.io/d/hXl0db

Printed it on 3.9 and it didn't help.

Checked M503 settings, and they were the same as at 3.8.1, seems firmware upgrade + factory reset with All Data option doesn't delete all data.

Downgraded back to 3.8.1, without factory reset, M503 showed default values - which was also expected after upgrade to 3.9 and factory reset.

Even though downgrade to 3.8.1 reset M503 settings it didn't reset Z-Level value, it stayed the same as I calibrated while on 3.9.

Printed gcode from Georgios on 3.8.1 - and it printed well it.

Note, I didn't do factory reset after downgrade to 3.8.1.

Belt status value stayed also the same as at 3.9. After selftest on 3.8.1 new values:

x: 275, y: 297. Will tension a bit Y belt to be around 280.

Funny thing is that Georgios noticed on my videos I shared with him that my side fan was mounted incorrectly - sticker was outside when it should be inside. Flipped fan.
```
Today I upgraded to 3.9.1 and had the same result, so I downgraded back. 

**To Reproduce**
I don't know how to help you to reproduce it, it's like only me having this issue.
Maybe I can send some technical logs from printer?
  
**Expected behavior**
Printer should print cones with the same quality on 3.9.* as on 3.8.1.
  
**G-code**
[cones_0.15mm_PETG_MK3S_4m (1).gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/5550561/cones_0.15mm_PETG_MK3S_4m.1.gcode.zip)

**Video**
  https://imgur.com/SYu0Ouh
on the top is result from 3.9.0, but it is the same for 3.9.1
and on the bottom is from 3.8.1
video for 3.9.0 https://photos.app.goo.gl/ms8H26EfXJdB87926
video for 3.8.1 https://photos.app.goo.gl/UDuFniE8BQikPY1M7
I didn't record 3.9.1 - it is the same as 3.9.0.

another view on cones: https://imgur.com/pnElVUp
video of temp tower print: https://imgur.com/sMZHW18

temp tower print on 3.9.0: https://imgur.com/WVIsNSW
temp tower on 3.8.1: https://imgur.com/kWQcG3t

DRracer, 2020-11-17:
@eliotik please change `M900 K0.12` to `M900 K0.07`, flash FW 3.9.1 and report back. Thank you

eliotik, 2020-11-17:
Hi @DRracer , I did next:
```
Send: M900
Recv: echo:Advance K=0.00
Recv:  E/D=Auto
Recv: ok
[...]
Send: M900 K0.07
Recv: echo:Advance K=0.07
Recv:  E/D=Auto
Recv: ok
```
Upgraded to 3.9.1, restarted  printer.
Tried to print same gcode file.
Here is a result: https://imgur.com/XHWDDce
It is worst than it was.
Now in pre-print step from nozzle comes a lot of filament, you can see in the beginning of video bulb of filament on the green tape.

DRracer, 2020-11-17:
@eliotik did you change the M900 line in the G-code file?
Did you make sure the nozzle was clean before starting the purge line? What I can see from the video looks very much like filament stuck on the nozzle during the mesh bed levelling phase and not an extrusion issue, but let's keep that aside for now.
I'll try printing your G-code from PETG as well to see if I get the same results.

eliotik, 2020-11-17:
@DRracer no, you didn't tell me to change it in gcode file, I set this command in printer settings.
Nozzle was cleaned before printing.
Why would thumbnails be an issue for my mk3s? :)
I will try to change gcode file and try to print it again and will record from the beginning.

DRracer, 2020-11-17:
I didn't expect you to send the M900 command separately and then print the unchanged file :) .

My result with K=0.07:
[image]
IMHO not bad considering my rather wet PETG and the nature of this stress test.

Nozzle: did you check the nozzle after the machine finished mesh bed levelling for any debree accumulated on the nozzle?

Thumbnails: it takes a lot of time to read those comment lines by the 8-bit board - you can get a watchdog reset. But probably the Octoprint strips these comment lines when feeding the code into the machine.

eliotik, 2020-11-17:
@DRracer I modified gcode file: 
[cones_k007_0.15mm_PETG_MK3S_4m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/5555534/cones_k007_0.15mm_PETG_MK3S_4m.gcode.zip)
It didn't help.
Here is a sequence of all video recordings to give you more details:
p1 - https://imgur.com/UyX0f4j
p2 - https://imgur.com/Bou8RHn
p3 - https://imgur.com/S7Sjqdr
p4 - https://imgur.com/L8cASGp

So you can see that on the nozzle there is a small tip, it should not be a problem, but close to the end of pre-extrusion line, line on 90% of it, it extrudes bunch of filament now which ends in this mess.

I can downgrade back to 3.8.1 and record it so you will see the difference.

Thank you for thumbnails detail, will stop use them 🙇

eliotik, 2020-11-18:
@DRracer I downgraded to 3.8.1 and without any other changes printed file, here is what I got:
p1 - https://imgur.com/oukxeyS
p2 - https://imgur.com/imNHzMY
p3 - https://imgur.com/1xnYDiD
p4  - https://imgur.com/P2YT0Lx
p5 - https://imgur.com/7QbOTiJ
photo of result: https://imgur.com/FMSm6cD

How to deal with that blob on the nozzle when it does pre-printing track? 🤣 

Anyway, what should I do with firmware or printer?

eliotik, 2020-11-18:
I tried to print again in 3.8.1 with putting on nozzle sock: https://imgur.com/PuCF1wB
During warm up, spaghetti was that long: https://imgur.com/dX5RiMl
It felt itself later.

Blob on pre-track - https://imgur.com/BhKcohA
Printing p1 - https://imgur.com/SPSoN9n
p2 - https://imgur.com/Xmd8BUV
Result photo: https://imgur.com/9b6Xkdo

Then I thought maybe it's a tape to which filament doesn't stick, I removed it and tried again and it sticked fine without blob. Sorry, I didn't record it this time.
So blob problem solved by removing tape, seems it became way to old 😄 

But firmware problem still exists.

What kind of debugging info can I provide?
Can I somehow create a snapshot of my firmware with data and settings and upload to you?

DRracer, 2020-11-18:
@eliotik hmmm, this is a surprising result. I still think it is a combination of K factor and the filament you have.
It would be interesting to find the correct K factor for this spool of yours though - please head to: https://www.prusaprinters.org/cs/prints/39968-la15-k-factor-calibration and download _0.2mm | 45mm/s QUALITY_ G-code for PETG. 
Print it (takes a few minutes) with your orange PETG and post the photo.

Anyway - what is your nozzle size? Any upgrades on your machine (like Bondtech geared extruder)?

eliotik, 2020-11-18:
Hi @DRracer, I didn't do any upgrades to extruder beside fan cover and stepper visualizer.
Here is result with 3.8.1: https://imgur.com/pKGlzaA
and for 3.9.1: https://imgur.com/PGrHgpF
I looked at them and I cannot tell a difference.
Filament is Prusament PETG Orange. I have the same issue with Prusament PETG White and all Overture filaments and all NEAT filaments. I tried both PLA and PETG, bright and dark colours. 
Nozzle size is 0.4

DRracer, 2020-11-18:
@eliotik wow, so the Linear Advance K factor is correct (look at the pattern printed with 3.9.1 and the most consistent line is the one you want = 0.08 or so), your printer is completely stock (at least in terms of extrusion and related stuff), you are using well known filaments, so why does it make such an extreme overextrusion? 

How old is your nozzle, i.e. how much filament did it go through? I'd say it is relatively in shape due to the fact, that the optimal K factor came out around 0.08. Worn nozzles need lower K factor.

I'm running out of ideas especially when my machine printed the part relatively ok. We must find a way to replicate your issue.

eliotik, 2020-11-18:
@DRracer I'm using this nozzle from e3d: V6 Nozzles - Nozzle X / 1.75mm / 0.4mm
But it's the same with stock nozzle which came with the printer.

Is there a way to snapshot firmware or motherboard data and sent it to you?

wavexx, 2020-11-18:
@eliotik Do you have a 3mf file for the model already cut?

wavexx, 2020-11-18:
I tried printing both at K0.12 and 0.08. There's some difference, but it's only noticeable on the top layer looking closely:

[image]
[image]

(poor light resulting in fuzz, sorry about that). There's some overheating going on at the tip, but the rest looks good. Nothing remotely comparable...

@eliotik can you try to print the following gcode using the FW 3.9.1:
[cones_nola_0.15mm_PETG_MK3S_4m.zip](https://github.com/prusa3d/Prusa-Firmware/files/5563343/cones_nola_0.15mm_PETG_MK3S_4m.zip)

It's the same, but with linear advance disabled. I tested this here, and the result is again almost indistinguishable on the cones.

Your images somehow remind me of the overextrusion going on in some prints in beta versions before 3.9.1, but amplified :thinking:

eliotik, 2020-11-19:
Hi @wavexx , I believe this is 3mf file I used to create gcode: [cones.3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/5563445/cones.3mf.zip)

I tried to print your gcode, and it worked, difference with 3.8.1 is that it has more stringing and at about 85-90% level it had some weird slowdown, but than speed restored.

Here is a photo of result: https://imgur.com/R1iaWmD
And video recordings:
p1 - https://imgur.com/fW419dS
p2 - https://imgur.com/zYBg5dM
p3 - https://imgur.com/B9zYQRW

offtopic: on your images, is it printing plate? if yes, how come you don't have a single scratch on it? 😄

DRracer, 2020-11-19:
@eliotik so setting K=0 (Linear Advance off) did help a bit. So it is really related to the extruder motor (stepping or something mechanical). Can you please verify your bondtech gear is fixed tight to the motor's shaft?

Print sheet: how did you manage to scratch it so much :) ? Honestly, I'd really like to know. I also don't have a single scratch even though I'm using a flat knife (like a spatula) to remove prints.

wavexx, 2020-11-19:
> I tried to print your gcode, and it worked, difference with 3.8.1 is that it has more stringing and at about 85-90% level it had some weird slowdown, but than speed restored.

For later reference, the slowdown is here:

> p2 - https://imgur.com/zYBg5dM

Ok, I'm not sure why I don't get either the slowdown and also the prints with LA enabled. The printing speed is too slow to be due to the internal planner.

I know it's annoying helping to find the issue, but could you try to install FW 3.9.1 and do a full-reset (hold the button just after the reset, then select erase all). This will require doing again the calibration I'm afraid, but you can note down your current Live Z offset (the new one should be *very* close). Then print your original gcode again.

This is to clear my doubt that an existing setting previously saved (XYZ calibration mostly, but some other setting like microstep maybe?) could influence this issue.

> offtopic: on your images, is it printing plate? if yes, how come you don't have a single scratch on it? smile

Yup. If flexing the sheet is not enough, I generally pry the pieces with a blade on the printed part to lift off a corner, so I don't use a spatula. But, I have another sheet that I destroyed with a screwdriver, and a hole I made printing PETG... so eeeeh.

eliotik, 2020-11-21:
Hi @wavexx , sorry for silence, I just did a factory reset->all data, calibrated through wizard and set Live Z value.
After that I tried to print from SD - it didn't work, tried to print from Octo and again the same bad result.
Do you have any other ideas I could try?

closed 2023-11-02 by github-actions[bot]

eliotik, 2024-04-02:
I'd like to share a quick update, I was skipping FW updates for 3 years, and in December'23 I gave it a try and installed recent at that time FW version, I have no problems, everything prints as expected.