prusa3d/Prusa-Firmware #2898: Bed Thermal Heating Runaway failure

taz43mca, 2020-11-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK2.5S, MK2.5,
**Printer firmware version**-  3.9.1

**MMU Upgrade** - NA
**MMU upgrade firmware version - NA

**SD card or USB - both
  Please let us know if you print via SD card or USB- both

**Describe the bug**
Full description of problem and steps taken to resolve are posted on the PRUSAPRINTERS.ORG Forum under **English Forum >Original Prusa i3 MK2.5S & MK2.5> Hardware, firmware and software help> Bed Thermal runaway**   
Here is the short version:
About a six months ago I had a Bed Thermal Runaway error on my printer. After exhausting tests and conversations with PRUSA and on the forum, I replaced my mini-Rambo 1.3a board with a spare. That cleared the problem. This is after 2 years of no problem printing.
A few weeks ago I had the same problem. A full description of the problem and action taken are posted on the forum ass discussed above. Some of the more significant information is posted below. Some is repetative since I did a copy paste from the forum. Best to read the chain of events on the forum page:

"Well it is back again. LCD screen shows the correct bed temperature (same as before). Same printer as before (my MK2.5S).
Here is what happened: I was printing a 4 color print with a filament change in between each color (do not have an MMU). I had printed this 2 or three times with no problem. Well this time I left the printer and it stopped while I was away from it (for a color change). When I came back after an hour or so, the printer was beeping (normal) BUT the screen gave me the dreaded Bed Thermal Runaway. The bed temp and the estruder temp were back down to room temperature. I tried to get the print to restart but to no avail. Shut down printer and restarted but same error. So I raised the extruder temperature and removed the filament. 
Here are some readings:(taken after the event)
Bed has no light on when set to heat on LCD; Thermsister does react to heat;
With bed wires disconnected: wires read 3.39 M Ohms on Ohm meter while connected to Rambo; with meter leads reversed the wires read 20 K Ohms
The bed with wires disconnected reads .8 Ohms
Bed wires disconnected from bed and PSU OFF, I get .01 VDC and .015 VDC with meter wires reversed
Bed wires disconnected from bed and PSU ON, I get .063 VDC and .043 VDC
With all wires connected and printer set to heat on PLA, extruder heats up but no light on bed and I get 2.88 VDC and 2.90 VDC 
Heat-up ceases due to bed not heating. (at about 70C above the anbiant start temperature)

So, just for kicks, I took out the first board that had failed and took the same Ohm readings at the Bed 12 volt DC output connections on the Mini-Rambo 1.3a board. I fully expected to get the same or very similar readings.
Well, with the common multimeter lead connected to the positive pin, I did get the same 20 K Ohm reading that I got on the installed (failed board #2). Then I reversed the leads and put the common on the negative 12 Volt DC pin and the reading was 10 time what it is on the installed board, 34.6 Meg Ohms. Double checked both boards and got the same results.
All this is similar to what I had last time this happened and could not fix it until I replaced the Rambo with a spare I had.
All fuses are OK and so are all connections. PSU is working fine as the extruder has no problem warming up (until the software stops it due to no heating of the bed). Thermsister checks out OK also.

The bed LED is NOT on at any time which would indicate no power to the Bed or a burnt out light led. Multimeter measurement of DC power at bed connection point is zero with multimeter leads both normal and reversed. 

Preheat PLA 215/60 C on LCD selected. Z axis moved up a few mm, Bed temp 27C Extruder temperature went up to 99C with the following message on LCD "Err: MINTEMP BED" Z Axis moved up a few mm again when the ERR Message appeared, extruder had about a 5C drift up after Error message; #2 Bed 28C, Extruder at 79C to start. Z Axis moved up a few mm when preheat PLA selected then when extruder reached 139C received ERR message and z axis rise by a few mm and an extruder temperature drift up to 144C ; #3 Bed at 29C extruder at 34C, Z axis moves up, Err at extruder temperature of 106C, z axis moves up and temperature drifts to 110C; #4 Bed 28C, Extruder 50, z axis moves up at start, Err at extruder 117C, z axis moves up, extruder drifts to 122C; #5 Bed at 29C Extruder at 34C, z axis moves up at start, Err at extruder temperature of 105C, z axis moves up, extruder temperature drift to 111C
Then I did same PLA Preheat but at the start went to settings and set bed temperature to 27C, Extruder acted normally and went up to 215C.
Then I did a filament autoload but went into settings and set bed temperature to 27C with PLA at 215C. Filament Autoload worked normally as did extruder temperature at 215C (Actually autoload started about 200C)
Then I sliced a short print with PLA extruder set to 215C and bed set to 27C in Prusa slicer. Extruder went up to 215C and began print normally (bed started at 28C but with heat applied by printing drifted up 3C - no light on bed); second layer at 210C/27C and print finished normally.

Hair Dryer test of thermister showed a temperature rise in a normal manner. Tested from 27C to 40C.
NO BED Movement at time of event or anytime except when Bed Temperature lowered under Settings or in the slicer as described above.

Note that I did a reset of printer between some of the tests but saw no difference. Also, the difference of the final extruder temperature, at ERR point, appears to be about 70C so when I started with an extruder at 34C the final ERR temp was lower than when I started at 76C. The Bed temp was about 27C in all cases. This indicated to me that the ERR point was more of a temp difference from extruder start temperature  than a temperature difference between the extruder temperature and the bed temperature.

With bed set to 50C in settings and power on, bed output  pins read 2.76 VDC
with psu power off I get a reading of .160 VDC
with PSU off and power cable to PSU disconnected I get (after some maybe capacitor discharge) .056 VDC
heat bed examined and no defects
all fuses OK
PSU voltage to mini-Rambo bed connection is 12.6VDC

**Finally it was suggested that I do a factory reset and reflash the firmware -**
Guess what...that worked and the printer is working normally. It was after this that I realized that the first failure (first board replacement) happened after a firmware update (I think it was 2.8.0 or 2.8.1. May have nothing to do with it but thought you should know, The most recent failure happened shortly after the 2.9.1 FW update.

Both FW updates for my MK3S appear to be OK as I have had no problems with that printer.

I also suspect that the first board replacement was unnecessary as a reflash may have solved the problem in that instance so i will keep my first board as a spare.

Bottom line is that the REFLASH worked and all tests point to something happening with the firmware that caused the initial problem.

taz43mca, 2020-12-02:
Sorry to say I spoke too soon. I have been using my MK3S exclusively since the last failure of the MK2.5S, reflash and successful printing. Had the need for both printers today so started a print on the 2.5s and after a layer or so it failed with the same heatbed thermal runaway failure error message as before.  I can only think of two things to do and both cost a pretty penny: replace the Mini-Rambo 1.3a AND the bed. Any suggestions that would help are really appreciated. Posted this update on the PrusaPrinters.org Forum also.

hernandez87v, 2020-12-19:
Hey I had similar issues but in my case I have a MK2.5S on MK2.5 firmware. Reason being is I was having issues with the extruder for the MK2.5S and stayed with MK2.5 extruder. No issues whatsoever using prusa3d_fw_MK25_3_9_0-beta_3175_RAMBo13a_en-es.hex. 

Then I decided to upgrade to 3.9.1RC1 and this happened, https://github.com/prusa3d/Prusa-Firmware/issues/2822#issuecomment-705133742 got m112 error.

Now this does seem like regular wear and tear on the cables so I soldered them back and downgraded back to 3_9_0-beta_3175. No issues and running fine.

Yesterday I decided to upgrade again to prusa3d_fw_MK25_3_9_2_3524_RAMBo13a_en-es.hex and then again M112 error came back. So now instead of the resoldered end having issues my rambo board got fried a bit and now my heatbed seems too have  a short. It just seems like a really big coincidence, the first time I can pass as wear and tear but a second time M112? 

During troubleshooting the heatbed would turn on and connecter got REALLY hot on the rambo end and thats when I found it fried a bit. I've had this printer since 2017 and upgraded it although not fully running fine but the cost to fix it is a bit much now so ended up getting an Ender for now until I have the time and patiance to fix. It being my only source of income I needed something quick. 

I did speak with support and they helped as much as they could but ultimitely Its on me for hybrid MK2.5S with MK2.5 firmware. Just wanted to share my situation on the issue.

closed 2023-11-02 by github-actions[bot]