prusa3d/Prusa-Firmware #2899: [ENHANCEMENT]<Enter comprehensive title>

DylanBarber, 2020-11-20:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [e.g. MK3S, MK3], may also apply to other printers

**Is your enhancement related to a problem? -No
  
**Describe the solution you'd like**

PETG requires the nozzle to be further away from the bed than with PLA. Because of this I'm always adjusting my Z height (distance from the bed) when swapping between the two types of filament.

My enhancement suggestion is adding something in the menu, prior to printing, that allows you to save the Z height presets for multiple filaments and apply them BEFORE starting the print.

Suggested flow:

Menu -> Z Heights -> Menu displaying the following:
-----------
PETG: -0.580
PLA: -0.565
ABS: -0.575
ASA: -0.580
-----------
Then allow the user to select their Z height preset

Currently I'm just storing these values in a .txt file on my computer and then rushing to change it once my print starts but it would be so much more convenient if it existed in the firmware.

Thanks!

vintagepc, 2020-11-20:
You could do this with different sheet profiles for the same sheet - have you tried that feature?

StarWitch, 2020-11-22:
Would there be some way of detecting filament type based on GCode and then loading the "filament-type height offset" and applying it to the base Z adjustment? i.e:

Base Z Adjustment: -0.900
PLA: +/-0.00
PETG: +0.15
ASA: +0.10

etc, etc.

vintagepc, 2020-11-22:
I don't think they want to support that as it's a somewhat "risky" feature in terms of the printer damaging itself. It also potentially makes the g-code completely non-portable to other machines. 

https://github.com/prusa3d/Prusa-Firmware/pull/2201 might be of interest to you though.

StarWitch, 2020-11-22:
Oh, I didn't mean baking the offsets into the GCode itself, but rather just the filament type.
When you generate GCode with something like the MMU2S, you can see `filament_type` in the GCode as a comment listing the five filament types for each nozzle. One option would have the printer detect that `filament_type` for just a single value, and then apply the offset automatically based on settings on the printer itself. If a printer doesn't acknowledge the `filament_type` comment or if it's a filament type the printer doesn't recognize or have a preset for, then it just uses its own preset Z-height, no different than before.

I do like your PR #2201 though! Having it be a setting in PrusaSlicer or something like that would be just as good, I think, although it does run the risk of being non-portable if some other printer uses `M850` for its own purpose, but the risk of that collision is relatively low.

closed 2023-11-02 by github-actions[bot]