prusa3d/Prusa-Firmware #2904: [BUG] MMU2S Filament load failure

brettvitaz, 2020-11-29:
**Printer type**
MK3S

**Printer firmware version**
3.9.2

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
MMU2S

**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]
1.0.6

**SD card or USB/Octoprint**
Both SD and Octoprint are affected

**Describe the bug**
Extruder motor does not move after filament selection, causing failure to load selected filament. 
  
**To Reproduce**
1. Slice a single filament print
1. Start print on printer
1. At filament selection, select appropriate filament
  
**Expected behavior**
Pinter pre-heats hot end and loads filament successfully.

**Actual behavior**
Printer attempts to load filament, extruder does not respond likely due to cold extrusion protection, often failing to load filament.
  
**G-code**
[Shape-Cylinder_0.1mm_PLA_MK3SMMU2S_5m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/5612638/Shape-Cylinder_0.1mm_PLA_MK3SMMU2S_5m.gcode.zip)

brettvitaz, 2020-12-22:
`Tx` gcode should wait until a safe temperature before loading.

brettvitaz, 2021-03-14:
I have found this is a problem when using Octoprint and the plugin OctoPrint-Mmu2filamentselect. When the plugin times out and hands off the filament selection to the printer, the warm up is not respected.

gudnimg, 2022-10-03:
In 3.13 (currently in development) `Tx` gcode won’t require the nozzle to be preheated. The filament will be loaded into gears and stopped when the filament sensor triggers. `Tc` gcode loads the filament from the gears and into the nozzle, this step requires preheating.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg