prusa3d/Prusa-Firmware #2909: [BUG]PM702 (Filament unload) sometimes triggers M600 (filament change)

Its-my-right, 2020-12-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3s
**Printer firmware version**-  3.9.2

**MMU Upgrade** - No
**MMU upgrade firmware version - N/A

**SD card or USB/Octoprint**
  Octoprint

**Describe the bug**
  When I use M702 via octoprint (well, via a button mapped on octodash), sometimes it behaves well (unload and that's it), and sometimes, after the unload, the head is parking like for a filament change, and then ask to put in the new filament
  
**To Reproduce**
  Use M702 via octoprint. Sometimes it behaves as described
  
**Expected behavior**
  M702 should only unload filament everytime, and never ask as if a filament change was requested

JohnnyDeer, 2021-07-29:
Hi @Its-my-right, I can confirm that there is something wrong, thank you for reporting.

gudnimg, 2023-06-22:
Not sure if this was present in 3.12 and before, but this should definitely not be an issue in 3.13 anymore as filament runout is blocked.

I think we should consider closing this issue once 3.13 is out.

Prusa-Support, 2023-09-01:
It seems safe to close this issue as resolved.

If the issue persists via Octoprint (probably not tested), now the user can send `M406`/`M405` codes to disable/enable the filament sensor before/after the `M702` filament unload code (PR https://github.com/prusa3d/Prusa-Firmware/pull/1447).

Michele Moramarco
Prusa Research

closed 2023-09-01 by Prusa-Support