prusa3d/Prusa-Firmware #2913: [ENHANCEMENT] IR-sensor active during print (ON/OFF)

primuspingvin, 2020-12-09:
**Printer type**    MK3S 
*MMU Upgrade**  MMU2S
* code tested on: firmware 3.9.2

I have a suggestion for a very simple however important and useful addition to the firmware. In short, currently the IR-sensor does not detect failure DURING print. This is very bad when eg printing some fragile wood- or metalfills that sometimes *do* break in the quite sharp PTFE-bends at higher Z-levels (which means FINDA still thinks all is ok). Also, as a clogged hot end during print will introduce grinding, this would also get detected (as the IR-sensor will generate a signal when filament has been grinded away).

I think a menu item "IR active during print" ON/OFF would be great (a sort of emergency option for people with poorly calibrated IR-sensors)

The CODE CHANGE is VERY simple:

In  file mmu.cpp:
In  function  mmu_loop(void):
In  case S::GetFinda:
IN if (mmu_rx_ok() > 0):
(at line approx 389):

CHANGE: 
if ( !mmu_finda && CHECK_FSENSOR && fsensor_enabled)

TO:
if ( (!mmu_finda || !(PIN_GET(IR_SENSOR_PIN) == 0)) && CHECK_FSENSOR && fsensor_enabled)

(to switch it using a menu ON/OFF-item, I suggest just to tie the menu toggle variable to the IR-readout on the same line.)

Best regards, Hans

primuspingvin, 2020-12-09:
I realize the final unload will not work properly without a flag for when filament is available for printing. I added it as follows, and this code works for me so far. Then again, I do not know all details wrt to the code so I would be grateful if someone could double check the idea.

These are my changes to mmu.cpp:
1. add a flag:
bool activefilament = false;

2. Set flag true in void mmu_load_to_nozzle():
activefilament = true;

3. Set flag false in void extr_unload():
activefilament = false;

4. Change line 389 (approx, in mmu_loop(void) within case S::GetFinda:   ) to:
if ( (!mmu_finda && CHECK_FSENSOR && fsensor_enabled) ||
           (!(PIN_GET(IR_SENSOR_PIN) == 0)  && CHECK_FSENSOR && fsensor_enabled  && activefilament) { 

rgds / Hans

primuspingvin, 2020-12-10:
(I attached a tar.gz archive containing mmu.cpp, mmu.h and Marlin_main.cpp for fw3.9.2 in case someone would like to take a look at the full "mod"):

I added IR-sensor activation during multi-material prints as well as to the single material prints (I have done test prints for both cases). I would be grateful if someone would take a look at this improvement as I am not experienced enough to push something complete on my own.

For single-material prints, the Prusa firmware handles the filament movement within an internal function. However, as this is not the case for mmu-prints (a sequence of gcodes in the print file does the job), the IR-sensor will signal "no filament" during unload *before* the Tn happens. 

I see various solutions to this "unloading detection issue":
* modification within the slicer so gcode does not move filament beyond the IR-sensor before the Tn
* adding new gcodes for turning on and off the IR-detection (these codes can then hopefully be added to the custom gcode section of the slicer)
* Decide to turn off IR-detection based on a large negative extruder motor movement within the G1-code.
* Move the load/unload code into firmware (re-use existing functions already there)
* (sneak preview what's happening upstreams in the gcode-file; this I do not consider a realistic option)

I implemented the simple prediction within the G1-code, where the filament sensor is deactivated whenever more than 10mm of filament is unloaded (larger negative move than E-10). I would be more than happy if someone would improve or replace that solution.

To add the IR-sensor for mmu prints, in addition to the first 4 changes I already submitted, I added the following changes 

5. to mmu.cpp, in mmu_get_response(), in case MMU_UNLOAD_MOVE:
activefilament = false;

6. to mmu.cpp, in mmu_get_response(), in case MMU_TCODE_MOVE:
activefilament = false;

7. to mmu.cpp, in mmu_continue_loading(), directly after if (success) success = can_load(); * two occurences *
if (success) { activefilament = true; }

8. to mmu.h:
extern bool activefilament;

9. to Marlin_main.cpp, in get_coordinates(), directly after the line "if (i == E_AXIS) {":
if  (destination[E_AXIS] < -10.0) {activefilament = false; }

Most noticeable is that this code *already* (after using it only for two days) saved a print for me!!! FormFuturas "willow" woodfill filament snapped within the orange PTFE tube last night and the printer paused just as I hoped for. What a relief!

rgds, Hans

(BTW thanks to Mikael for ideas and support)

[IRsensor_active_during_print.tar.gz](https://github.com/prusa3d/Prusa-Firmware/files/5675986/IRsensor_active_during_print.tar.gz)

Auge103, 2020-12-13:
If this makes it into production, please add an option to turn it off. 
Printing flex filaments will most of the times not trigger the MMU sensor and it will fail to start the print despite having loaded the filament - you can currently circumvent this by triggering the filament sensor by hand when the filament touches the gear and it will start the print.

primuspingvin, 2020-12-13:
> Printing flex filaments will most of the times not trigger the MMU sensor and it will fail to start the print 

I agree it should be an on/off option (I suggested it in the title, however not myself experienced enough to write the menu item reliably). Said so, The sensor on my mk3s/mmu2s works flawlessly even with 70A - I know it is often a matter of very precise calibration (Prusas advice is not good enough for calibration in these cases). 

In case you are interested, I would like to recommend the following IR-sensor enhancement:
https://www.thingiverse.com/thing:3554066
I use it since long time (I even designed it, actually) and it has become quite popular. It might be worth a try in case the filament sensor bothers you for flex?

If you are into flex, I also suggest promoting the following pull request made by Mikael:
https://github.com/prusa3d/Prusa-Firmware/pull/2914
The mod makes it possible to slow down the T<n> loading phase (including the MMU), reducing flex loading errors to a minimum (very handy).

3d-gussner, 2020-12-14:
@primuspingvin Thanks for the idea and the code you provided.
I tried to implement it in the firmware and created a new branch you might continue working on https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_MMU2_IR_active

- The first tests have been good
- I can still see some issues that need some attention and polishing.

Here some tips for building / testing and providing your code changes.

- Use **git**
  - Read a bit how to use **git**
  - Create a **fork** of the Prusa-firmware to your github account via Github web interface
  - Open terminal
  - Use `git clone https://github.com/primuspingvin/Prusa-Firmware.git /Prusa-Firmware/MK3` to create a local copy of your github repository
  - `cd Prusa-Firmware/MK3`
  - `git remote -v`  shows you which repositories you are "watching" for updates
  - Use `git remote add upstream https://github.com/prusa3d/Prusa-Firmware.git` to add Prusa-Firmware repository to look for new updates
  - Use `Git remote add 3d-gusser https://github.com/3d-gussner/Prusa-Firmware.git` to add my git repository to look for new updates 
  - Use `git fetch --all` to check for updates
  - Use `git worktree` or `git branch` (I prefer worktrees as it is easier for me to see changes using tools like Meld/WinMerge)
  - Create a new worktree for your idea
    - while you are in the "Prusa-Firmware/MK3" folder and branch use `git worktree add -b MK3-<your_idea_name> ../MK3-<your_idea_name>`
      - this will create a "copy" of the "MK3" branch as "MK3-<your_idea_name>" branch in the folder "Prusa-Firmware/MK3-<your_idea_name>"
    - Use `cd ../MK3-<your_idea_name` to switch the folder and branch
    - Change the code to your needs
  - `git branch` will show you which branches you have and which is active `*`
  - `git commit` will commit you changes
  - `git push`  will push your committed changes to you repository
  - `git merge` will merge changes from other branches to your ( there can be conflicts ... read bit more about i)

- Use `PF-build.sh`
  - It will prepare a known stable build environment for Prusa-Firmware
  - It will build the firmware

- Use MK404 to test firmware https://github.com/vintagepc/MK404
  - especially testing MMU2S changes can be very very time consuming on real hardware.

- When you have tested and polished the code you can create a Pull Request

A PR is way easier to be reviewed than having code in an issue and somebody picking it up to implement it.

Again thanks for the idea and already done work, looking forward for the improvements.+

Appreciate you took your time not only to report the idea but also to work on the solution. Maybe this short "how to" will help you to provide the code as a Pull Request.

primuspingvin, 2020-12-15:
> I tried to implement it in the firmware and created a new branch you might continue working on https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_MMU2_IR_active

3d-gussner, thanks very much for your help. If I may ask you for a quick favour, I just added a branch handling a DIFFERENT issue, as it is easier to verify and therefore start with as my first possible future pull request. I would appreciate much if you could just briefly tell me if that branch follows up to standards? (I successfully used ./PF-build.sh for English only, MK3S, and have verified the idea is working on my MK3S/MMU2S). What more am I expected to do before it would make sense to request a pull?

https://github.com/primuspingvin/Prusa-Firmware/tree/MK3_can_load_with_IRcheck

It adds an extra check to can_load() to avoid/reduce failed prints due to the fact can_load() sometimes fails (I have seen the following happen many times):

Sometimes, the filament CANNOT be inserted properly, just to the beginning of the PTFE-tube where it gets stuck. However, when can_load() unloads the filament again (almost as long as the filament was loaded, it is unloaded again which makes up the check), the IR-sensor STILL detects the filament, likely either due to the filament got stuck, or due to thin strings fooling the IR-sensor.

My addition pulls the filament out another few mm after the first test and IF the IR-sensor STILL detects filament, apparently loading has failed! If success, then the MMU will load back the filament to the bond tech gears and continue as expected.

Furthermore, the addition hinders the MMU to unload filament if the IR-sensor still detects it when unload is supposed to begin, as it apparently got stuck and the MMU would only grind it anyway.

In short, instead of some failed prints, the printer will request user interaction. That's what I hope for. I have only used it for a few hours (wrote it this morning), so far all works as expected.

closed 2023-11-01 by github-actions[bot]