prusa3d/Prusa-Firmware #2916: [QUESTION]<How to log each G-code line read from the SD card to serial>

NickDGreg, 2020-12-12:
I would like to log each line of G-code to the serial port as it is processed.

Steps to achieve:
- the printer reads a file from the SD card
- each line it reads will be serial logged (this I can't figure out)
- those lines can then read via the serial monitor on a laptop (this I can do)

So by the end of the print, on my laptop I would have the reconstructed G-code file (plus whatever other logs the printer outputs).

I would like to achieve the logging by altering the firmware rather than adding an extra plugin/server like Octoprint (For understanding and experimenting purposes).

What I tried
I have looked in code and found [the print functions](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin.h#L99)  and examples of them in use in the code. This line is the "[command, which is to be excecuted right now](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/cmdqueue.h#L43)", but I think that would be the just one command not the full line.

The [cardreader](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/cardreader.cpp) or [SdBaseFile](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/SdBaseFile.cpp) are where I would expect a G-code line to be read such that I could add a print statement after it but I did not see where.

Would it be as easy as setting this [card.logging](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/cardreader.cpp#L30) bool to true? 

I imagine this is quite an easy thing to do, but I'm new to the codebase so am probably just a little overwhelmed. Any advice would be great! Thank :)

leptun, 2020-12-12:
You can look into the `CMDBUFFER_DEBUG` define. Enable it in cmdqueue.h and be amazed at the level of debugging that will smash the serial link. Of course, you will probably want to  tone it down a bit, so disable the debug parts of the code that are not needed.
With this define you should get when commands are added to the command queue, the source of these commands, when they are executed, when they are removed from the queue, (I think) when the queue is full and the number of commands left in the buffer.

I hope this is helpful

NickDGreg, 2020-12-12:
Yes that helped! Your tip led me to [this line](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin_main.cpp#L3696): `SERIAL_ECHO(cmdbuffer+bufindr+CMDHDRSIZE);` in `marlin_main.cpp` which does exactly what I wanted. 
(`CMDBUFFER_DEBUG` with the serial link open actually caused the printer to crash when it first reads the file)

closed 2020-12-12 by NickDGreg