prusa3d/Prusa-Firmware #2920: [BUG] MMU Load failure before homing can crash axis

ardenpm, 2020-12-16:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK2.5S with MMU2S
**Printer firmware version**-  3.9.2

**MMU Upgrade** - MMU2S (with new parts for 3.9.2 firmware)
**MMU upgrade firmware version** 1.0.6

**SD card or USB/Octoprint**
  SD but not applicable here since this is before printing or homing

**Describe the bug**
  If MMU loading fails before the printer has been homed it will attempt to move the axis (in particular the Y-axis) after failure to wait for user input, however since the axis have not been homed it may attempt to move beyond the travel limits.
  
**To Reproduce**
  The easiest way to reproduce is to remove the bowden between the MMU2 and the extruder, attempt to load a filament to the nozzle and wait for it to fail when it never reaches the IR sensor. Eventually the printer will detect the failure and move the axis to wait for user input.

In my case this attempted to move the Y axis beyond its travel limits crashing the bed.
  
**Expected behavior**
  Expected behavior would be that if the printer has not yet been homed an MMU error should not reposition the X/Y axis when waiting for user input.
  
**G-code**
  N/A.

JohnnyDeer, 2021-02-04:
Hi @ardenpm, not happening to me at 3.9.3, please try update FW and confirm this issue is fixed. Thank you

ardenpm, 2021-02-08:
I updated to 3.9.3 and unfortunately the issue still occurs. Are you using a MK2.5S?

gudnimg, 2023-07-12:
This is definately fixed in our 3.13 branch. Adding this to the milestone.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg