prusa3d/Prusa-Firmware #2933: [FEATURE REQUEST] Resonance Compensation

haarp, 2020-12-26:
Hello,

Klipper has recently introduced a very very cool feature called resonance compensation, which can massively reduce the ringing introduced by vibrations in the motion system. Details can be found here: https://www.klipper3d.org/Resonance_Compensation.html

It would be awesome to have this feature in the Prusas. Does the i3 have enough processing power left to do something like this? If one assumes that e.g. all MK3 have close to the same characteristics, predetermined values could have a great effect then.

Note that this is not the same feature as linearity correction or linear advance.

Thanks!

wakareva, 2021-03-26:
this could be a very nice implementation as fine-tuning to reduce gost and to print faster.

Hooch180, 2021-03-30:
After testing printer using accelerometer, you could theoretically apply the "correction" in the slicer. This doesn't need any more processing power on the printer end.

rickyroze, 2021-08-23:
Bumping this again since input shaping seems like a feature that could almost be baked in to the slicer.  It has been implemented in Klipper and would help reduce bands and ghosting.  It'd get rid of the ghosting that's even present on the printed parts of the Prusa mini and maybe even allow for a speed increase in print farm applications.  It'd be great to see a printer with native support for input shaping.

HopsDa, 2022-02-03:
Yes Please!

Baughn, 2022-06-14:
The difference is quite dramatic. After installing Klipper and setting up resonance compensation, using stock speed/acceleration settings, it went from approximately as noisy as a stock Mk3s always is, to whisper-quiet. On the same exact g-code, in fact.

I suspect it also has benefits for longevity, but even if I'm wrong, it'd still be a nice improvement.

As for me, I'm returning to stock firmware because the _rest_ of the ecosystem is better. Here's hoping I won't be without resonance compensation forever. Maybe it could be done as a standalone post-processor?

Prusa-Support, 2022-06-22:
Hello! Thank you for your posts and for the feedback on this aspect. I'll bring this up to our devs for them to further evaluate this aspect. In case, we can further check the possible noises and resonances with stock firmware in detail with me info@prusa3d.com mentioning me, beyond the compensation, we might identify possible hardware contributor causes, although it's relatively hard to pinpoint it hardware-wise. Or, feel free to send us some video comparisons of the resonances even here on this thread. 

Alessandro Pantaleo
Prusa Research

peterduniho, 2022-10-22:
From looking at online resources discussing input shaping, it appears that there are a few different ways to approach implementation. One example I found showed how to print a test piece, using a high acceleration rate to ensure ringing shows up on the print, then to measure the distance between ringing features, and finally use that distance in a formula to determine the peak resonance frequency that needs to be removed.

It's not clear to me whether this can in fact be implemented in the slicer, but if it can, that would obviously be a great approach for the Prusa ecosystem, since that's the easiest for users to upgrade. But a firmware implementation would be great too. For me, the most important thing is a) get it done, and b) come up with an implementation that doesn't require additional hardware (e.g. accelerometer), so that it can be deployed on my existing MK3S+ with just a download and some simple calibration steps.

Mike8-I, 2022-10-23:
If the 8bit processor has enough umph to do the calculations and/or someone comes up with a clever algorithm, I would prefer a firmware solution. But a slicer solution would do, too. Anyway, to keep the MK3 competitive, this looks more and more like a "must have".

Prusa-Support, 2022-11-24:
Hello, thank you for the additional comments on this thread - unfortunately, I wouldn't be able to personally address the specific points, but I passed everything on to our devs, they are evaluating the possibilities for this. 

Alessandro Pantaleo
Prusa Research

3DCoded, 2023-01-09:
Any updates on this? Marlin Firmware has had this since 2020.

leptun, 2023-01-09:
> Any updates on this? Marlin Firmware has had this since 2020.

Given that input shaping requires a lot of ram and flash which we don't have on the atmega2560, most likely it will never be implemented. Not in this firmware anyway. Even marlin struggles to run input shaping on the AVR. Klipper is still the way to go if you want resonance compensation on the 8-bit Prusa printers.

> As for me, I'm returning to stock firmware because the rest of the ecosystem is better.

Yes, the Prusa-Firmware is considerably more user friendly than klipper, but it is quite EOL at this point given the current hardware limitations. Klipper is still the way to go for any fancy new features since it has the advantage of offloading a lot of processing to the host CPU and isn't constrained by 8KB of RAM. That being said, the klipper experience can still be improved (especially the UI) if you set up everything right. I might share at some point my MK2.5S/MK3S klipper configs which I've been using for quite a long time. If anyone is interested in the configs, please let me know.

Mike8-I, 2023-01-21:
I would like to offer a idea that might offer better looking prints without loosing (maybe even gaining) print speed:
The root cause for the ringing are resonances that are excited by harsh accelerations or jerks ( visible on the outer perimeter).
Instead of compensating the motions of the resonance (fixing symptoms) we could avoid to excite the resonances (fixing closer to the root cause).
This can be done in the slicer. Before reaching a corner, the slicer decelerates with an adjustable rate and then also accelerates with this same rate after the corner. This could be tested with only the outer perimeter or all perimeters. The ( adjustable ) corner-acceleration that works  could be somewhere in the range 10 to 100mm/s x s.
As this does not influence infill, it would not do to much damage regarding print time. I could even be possible, that faster prints are possible, because the perimeter speed can be chosen higher. The ringing would not be affected by the higher speed, as the corner acelleration would still be small, just the acceleration distance would be longer.

Prusa-Support, 2023-01-31:
Hello! Thank you for your idea, though I wouldn't be able to promise anything at this stage, feel free to open a new issue on the PrusaSlicer GitHub section, unless there might have been something similar - so that the PrusaSlicer devs can take a look and evaluate the possibility! https://github.com/prusa3d/PrusaSlicer/issues 

Alessandro Pantaleo
Prusa Research

peterduniho, 2023-10-26:
As I understand it, PrusaSlicer 2.6.1 has input shaping support for the MK4 printer, and future support for MK3.5 with the xBuddy board. These either completely address the request here, or at least suggest a path forward for input shaping support on other MK3 models. For example, parameterized shaping where the parameter values are determined through specially designed test prints.

So it seems that either this issue can be closed, or updated with more detail about plans in the near future to address it.

peterduniho, 2023-10-26:
To be clear, you mean without the xBuddy board upgrade, which I assume replaces the 8-bit system with one capable of running the input shaping support?

That's how I read the statement here: https://blog.prusa3d.com/announcing-original-prusa-mk4_76585/

> "if you upgrade your MK3S+ with the xBuddy board to MK3.5, you’ll be able to use Input Shaper and Pressure Advance, too, with a future firmware update"

gudnimg, 2023-10-26:
You’re correct the xBuddy board is required.

arekm, 2023-10-26:
Maybe just close this issue as not planned (since this issue is about 8bit firmware only).

closed 2023-11-01 by Prusa-Support