prusa3d/Prusa-Firmware #2938: [ENHANCEMENT] Allow to change z adjustment whilst preparing to print

jokohoko, 2020-12-31:
**Printer type** - All

**Is your enhancement related to a problem? Please describe.**
When I have started a print and notice the first layer wasn't low enough (so some parts came loose of just printed with gaps), i cancel the print and restart it. Often forgetting to adjust the live z height first.

I know i want to lower it with about 0.025 mm. So i would like te be able to go to settings and lower it a little without having to wait for the heatup and first layer calibration to complete. At the moment the Z height adjustment is not in the menu before the print starts.

amarilio, 2021-01-15:
To add to this - long pressing the knob to change the z height also doesn't work before the print started, it'd be great to have it available in the preheating and calibration stages.

alexwhittemore, 2022-01-20:
Seconded - I'd love the option to press and hold during warm up etc to at least see the live Z value. It'd make the whole process of dialing in the first layer a little less hectic.

closed 2023-11-02 by github-actions[bot]