prusa3d/Prusa-Firmware #2939: [BUG] Many (10+) crashes in Y-axis per hour after performing new calibration method for MK3S with 3.9.3

scottmudge, 2021-01-01:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S+
**Printer firmware version**-  3.9.3-rc1
**Using SD Card**

**Describe the bug**
  I flashed the new firmware using avrdude, as normally done, which was successful. I then recalibrated using the full Wizard sequence, to use the new XYZ calibration method for the new PINDA probe. It took quite a bit longer, as it took many sample points. But it completed successfully and said everything was perpendicular.

However, I printed a basic (very little curvature) 2-hour print, and encountered over a dozen crashes in the y-axis. They recovered fine, but the print quality was much poorer than before (more zits). A crash in any axis has never happened before in 24+ hour prints in the past with the previous calibration method, not even a single crash. No changes to the hardware, it's been sitting in the exact same position, location, temperature, humidity, etc. Only change was the firmware.
  
**To Reproduce**
  See above.
  
**Expected behavior**
  No crashes in the firmware during printing.
  
**G-code**
  Attached.
[PC_Case_Front_Plate_v1_0_0.2mm_PETG_MK3S_2h21m.zip](https://github.com/prusa3d/Prusa-Firmware/files/5758625/PC_Case_Front_Plate_v1_0_0.2mm_PETG_MK3S_2h21m.zip)

scottmudge, 2021-01-01:
Following up to note that I re-flashed 3.9.2, reset the XYZ calibration, re-ran the wizard to recalibrate, and printed the same object.

I encountered zero crashes this time. I don't have any UART logs from the print with all the crashes, but it does seem related to 3.9.3.

espr14, 2021-01-04:
Thank you for the report. This is very weird since the only changes were in the calibration measurement algorithms. This cannot influence the printing itself. However, the output does influence the printing. Can you send me all the 5 numbers from the "Support -> XYZ cal.details" menu? Could you try a print with FW 3.9.2 calibrated with 3.9.3 and vice versa (FW 3.9.3 with 3.9.2 calibration)?

scottmudge, 2021-01-05:
Hello,

I have recompiled 3.9.3-rc1 and encountered the issue again. 3.9.2 (re-calibrated with 3.9.2.) did not crash on the y-axis a single time. But 3.9.3-rc1 (calibrated with 3.9.3-rc1) did crash on the y-axis. It is reliable on my machine.

## Data from 3.9.3-rc1 calibration:

**Bed Profile:**
```
Send: G81
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   -0.08417  0.12250  0.26639  0.34750  0.36583  0.32139  0.21417
Recv:   0.04139  0.23052  0.36019  0.43037  0.44108  0.39231  0.28407
Recv:   0.12722  0.30019  0.41670  0.47676  0.48037  0.42753  0.31824
Recv:   0.17333  0.33148  0.43593  0.48667  0.48370  0.42704  0.31667
Recv:   0.17972  0.32441  0.41787  0.46009  0.45108  0.39083  0.27935
Recv:   0.14639  0.27898  0.36253  0.39704  0.38250  0.31892  0.20630
Recv:   0.07333  0.19519  0.26991  0.29750  0.27796  0.21130  0.09750
```

**XYZ cal.details:**
```
Dist From Min (Y):
    Left: 11.02mm
    Right: 10.93 mm

Measured Skew: 0.0 deg

Point Offset: 
    X: -0.02 mm
    Y: 0.53 mm
```

I also printed the mesh bed level validation model (large, flat patches, single-layer), and noticed the 3.9.3-rc1 calibration is far worse. The tip was too low for all 4 corners (had to live-adjust Z +0.2mm), but it was too high for all areas in-between (no adhesion). 

This was not a problem with 3.9.2.

I will re-flash with 3.9.2 and see if the crashes still occur, to validate if it is the calibration data itself, or some other part of the code (which I don't think it is, as you noted there were no code changes that could have caused this).

My gut instinct now is that the 3.9.3-rc1 calibration values are just sub-optimal.

scottmudge, 2021-01-05:
Update:

3.9.2 with _**calibration from **_ 3.9.3-rc1 (same calibration as in post above) **also** crashes in the Y-axis.

It appears related to the calibration produced *by* 3.9.3-rc1, as expected.

I will recalibrate with 3.9.2 and see if the crashes continue to occur.

scottmudge, 2021-01-05:
Update:

## Data from 3.9.2 calibration:

**Bed Profile:**
```
Send: G81
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   -0.00750  0.06972  0.13111  0.17667  0.20639  0.22028  0.21833
Recv:   0.00861  0.07653  0.12805  0.16315  0.18184  0.18413  0.17000
Recv:   0.02861  0.08592  0.12743  0.15315  0.16308  0.15721  0.13556
Recv:   0.05250  0.09787  0.12926  0.14667  0.15009  0.13954  0.11500
Recv:   0.08028  0.11240  0.13354  0.14370  0.14289  0.13110  0.10833
Recv:   0.11194  0.12950  0.14027  0.14426  0.14147  0.13190  0.11556
Recv:   0.14750  0.14917  0.14944  0.14833  0.14583  0.14194  0.13667
```

**XYZ cal.details:**
```
Dist From Min (Y):
    Left: 10.69mm
    Right: 10.49 mm

Measured Skew: 0.08 deg

Point Offset: 
    X: -0.05 mm
    Y: 0.20 mm
```

The only significant difference I see is the Y point-offset.

I will print again to validate if a crash occurs in 3.9.2 with this new calibration.

espr14, 2021-01-05:
Great output. MBL should not be influenced at all since this calibrates Z height. XYZ calibrates XY and check correct height of Z only. But MBL is influenced by a temperature of nozzle and bed. SuperPINDA itself is not influenced but the bed changes shape and the plastic parts in the extruder do as well. To get consistent results, long preheat is recommended.
Could you connect the printer to the USB and save all the data sent during the XYZ calibration with FW 3.9.3 RC1? That would be really helpful.
I'm afraid of the 0.00° skew. That means you have totally perpendicular axes. On the other hand, such a small number (0.00xxx) can lead to numerical instability. The issue could be deep in Marlin.

scottmudge, 2021-01-05:
Indeed, it may be the skew of the 3.9.3 calibration. It is routinely non-zero in 3.9.2, but always close to 0 degrees in 3.9.3.

I also wanted to note that the crash indicator values may not have been correct in the latest 3.9.3 print. It's been reporting "1" in the y-axis for a while, but the total has not increased in a few prints. I'm betting the number simply isn't resetting to 0. Perhaps because I am using OctoPrint for printing lately.

I'll keep an eye on this and report back with the calibration output from 3.9.3, but unless others are reporting this issue, it looks like it may be isolated.

Hooch180, 2021-01-16:
I do have random Y axis crashes.

Firmware: 3.9.3.3556

This is my 2nd prusa and 4th printer overall I assembled.
My printer prints perfectly. Axis all move very smooth. Belt tension is correct as well as number reported is 260.
Steppers are cold as I added 2 or 3 (40mm x 40mm) heatsinks to all of them.

I get about 2 crashes per hour of printing. At totally random locations.

AnhDuck, 2021-01-17:
I ran a 6 hour print after updating to this firmware version and my machine experienced 8 crashes in total. All were in the y-axis too.

scottmudge, 2021-01-17:
I wanted to note that printing over USB w/ OctoPrint does not cause crashes, but it may be due to a lack of effective crash detection? In any case, it doesn't seem to be a physical fault, as the prints come out just fine.

Hooch180, 2021-01-18:
> I wanted to note that printing over USB w/ OctoPrint does not cause crashes, but it may be due to a lack of effective crash detection? In any case, it doesn't seem to be a physical fault, as the prints come out just fine.

It is hard to say why it could do it. The difference is only source of commands. You can always start print from OctoPrint and try stopping Y axis with hand to see if it detects crash. (It should).

I'm printing from SD as I noticed that Octoprint is too slow. It slows down when there are many short instructions (like printing detailed arcs). Serial Port on Einsy board is just too slow.

Naugrimohtar, 2021-01-19:
I too am having this issue

rockybulwinkle, 2021-02-23:
Yup, also having this issue on latest firmware.

dvangestel, 2021-03-21:
I assembled my new printer yesterday, and I got at least 2 crashed while printing benchy. I only noticed them in the last 15 minutes of the print, there could have been more earlier.

rockybulwinkle, 2021-03-21:
There is a fail stats somewhere in the settings. I forget exactly where it
is in the menus. I gave up on normal mode and use stealth exclusively even
for multi day prints and haven't had any problems.

stleitner, 2021-04-20:
"Fail stats" is a top level menu, "Last print" sub menu shows the last/current failures.

In 3.10.0-RC2 the problem is not solved, yet. In the last 10 hour print I had 7 Y crashes (188 total since February). In most cases it recovers, but sometimes it is visible, if the filament is burned or the layer shifts

rockybulwinkle, 2021-05-25:
Has anyone checked if the release version of 3.10.0 has fixed it?

espr14, 2021-05-27:
Could you try the latest FW with factory reset without running XYZ cal.?

scottmudge, 2021-05-31:
Reflashed 3.10.0 and recalibrated after complete teardown and rebuild of my printer. I made sure the y-axis is perfectly assembled, tensioned, etc. 

Still encountering the y-axis crashes if I calibrate with anything after 3.9.2 (3.9.3 and 3.10.0 seem to cause this issue). 

I've briefly looked at a diff between 3.9.2 and 3.9.3. Nothing stands out which might be to blame beyond changes in calibration, but it doesn't make sense issues would be limited to the y axis.

And again to confirm, the printer itself appears functionally correct. If I disable error checking, it prints without error. But something appears to be falsely triggering the error detection.

I will take a deeper look at this and debug it more aggressively later, but for the time being it doesn't appear to appreciably affect performance.

stleitner, 2021-05-31:
I had a y-axis crash after the 3.10.0 update again. I could reproduce this crash with the bundled Triceratops head [gcode](https://www.prusaprinters.org/prints/3114) at about 26 mm z height every time. Anybody else had a y-axis crash with the bundled gcodes?

scottmudge, 2021-06-01:
Even after recalibrating with 3.9.2 before switching to 3.10.0 I'm having frequent crashes on the y-axis still. So in my case it no longer appears related to the calibration. It may have simply been coincidental in the past.

I thought it might be due to overtemperature conditions after having put my printer in an enclosure, but the ambient temperature inside never exceeds 35C. It also doesn't look like the overtemperature diagnostic bits are enabled in the TMC2130 `GCONF` register.

I'm at a loss. Belt test is perfect: -- 265 on both X and Y. Y-axis stage runs smoothly, it's been lubricated.

If it's related to the code, it's something that doesn't look immediately related to the drivers or stepper control. 

My next step is to rule out the g-code file. It does seem *reliable*, in that it crashes at roughly the same layer every time.

scottmudge, 2021-06-03:
An interesting update:

After setting the `SFILT` bit to 1 on the `COOLCONF` configuration register for the TMC2130 of the **y-axis** only (the one encountering the crashes), the crashing stopped. 

It looks like it was enabled globally for every axis at one point in `tmc2130.cpp`, in the `tmc2130_home_exit()` method, but it was commented out. It's probably best not to enable it for every axis, but setting it for the y-axis did seem to solve my problems.

For those unaware, according to the [TMC2130 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf), the `SFILT` bit enables filtering on the `stallguard` input, averaging all 4 phases/steps instead of using higher resolution. My assumption is that (at least for my particular motor), one of the coils/phases/steps is producing noise sufficient enough to falsely trigger the stallguard feature. By averaging it out with the filtering, it presumable reduces the chance of false triggers.

I also bumped up the failure count threshold from `64` to `96` on the y-axis, but when I did this without enabling the `SFILT` bit, it still triggered crashes. So something about the stallguard filtering has corrected my problem.

Perhaps other developers with more experience using this stallguard filter know better about it's drawbacks, but at least for me, it solved the issue.

leptun, 2021-06-10:
@scottmudge It could be that the threshold is just a tad too sensitive. Lowering the SG threshold by 1 on the Y axis could be enough to stop these crashes.
It would be interesting if this SFILT bit was active during printing, but disabled during homing. This way the homing stays precise, while spikes in the SG output are filtered while printing.

@DRracer what do you think about this?

scottmudge, 2021-06-10:
Admittedly I'm just a single data point, but adding a "low sensitivity" mode in the error detection setting might be a compromise. This may also help differentially diagnose other issues -- e.g., if the crash rate is reduced in low sensitivity mode vs. the same crash rate in both low sensitivity and normal modes. 

On the other hand, it may also defeat the purpose of error detection if the sensitivity is lowered or threshold is increased too much.

Just something to think about. It's strange everyone is associating it with 3.9.3/3.10.0, since the git history for the TMC2130 source code shows no changes in these releases, but I honestly can't find any other potential cause. 

But confirmation and attribution biases may just be causing people to conclude causation where it's only a correlation. I've replaced my bearings and everything else I can think of, but it's still possible the driver itself is going bad, or there is something like a dry solder joint. And I just assumed it was something in the release, when it was, in fact, something else.

scottmudge, 2021-07-04:
Short update, I've recently replaced my main board and the crashing resolved, even without modifications to the driver configurations I had noted earlier.

So, at least in my case, it looks like a driver or perhaps some discrete component had gone bad, coincident with being around the same time I updated firmware.

leptun, 2021-07-04:
Could the connectors be at fault here? Maybe one of the crimps not making perfect contact or some oxidation on the connector itself?

scottmudge, 2021-07-04:
I did think about that, but I had replaced the motor twice before the main board, and each time I replaced the motor I also replaced the entire 4-conductor motor wire with a shielded one to rule out outside interference, and also replaced the connector just to be safe. The same motor that was faulting before is working fine on the new board.

I just took a look at the conductor pins on the (presumably) faulty board under a microscope, and they look identical to the others, no visible oxidation or corrosion. The solder joints look fine, no cracking or anything. 

Not sure where the fault is, but unfortunately I do not have the appropriate logic/protocol analyzer hardware to verify if the driver itself was producing invalid output. I'm only making an observational assumption.

The only thing I can think of is that at one point something crashed in the firmware, possibly due to a low-power condition (brown out) during the night. This caused the printer to jam its y-axis motor continually against the end-stop for about 2 or 3 hours while I was sleeping. I'm not sure if the drivers are designed to continually push a NEMA17 to stall torque for 2 or 3 hours, but that may have caused damage.

leptun, 2021-07-04:
The drivers are smart enough to not destroy themselves (they enter fault mode). The run current on the drivers is also relatively low to what you would see on other tmc2130 boards (<=300mA). I think it's the printer mechanics that suffered the most during those two hours.

scottmudge, 2021-07-04:
That's what I figured, and thinking back the axis crashes occurred before that night anyway. But yes, the constant vibration did cause some loose screws. The belt looked okay though. It did fix a noisy bearing, so it wasn't a completely negative event. 

When I get some time I may probe the components around the driver more deeply, just to rule out a bad cap or perhaps a loose solder joint in the surrounding area. I may try to reflow the components around the driver, and the driver itself. A bad solder joint or loose connection does seem probable, given the symptoms.

stefan-am, 2022-03-11:
Printer type - MK3S
Printer firmware version- 3.10.1-4697

Hello,
could it be, that when printing from Octoprint, the counter for "Last Print Crash Y" is not reset? 

I printed the same part multiple times via Octoprint and observed a large number of "Crashes Y"  but very few -if any - for X. 
It showed for example 13 "Crash Y" after a 2h print. 
I was present during the entire print and recognized only the noise of 1 real XY-axis homing. 
Absolutely sure, there were no real 12 more crashes with homing on this print.

I did the identical print via SD Card and the counter showed "Crash X 0   Crash Y 1" after the print finished.

leptun, 2022-03-11:
When USB printing, the “last print” statiatics are indeed not reset. We are aware of this issue. You should only rely on the “total” statistics when usb printing. There is no immediate plan to fix this bug.

wavexx, 2022-03-12:
We do some hacks for the LA1.0 compatibility by abusing M81 as an "end of print" marker, which is horrible. In this case though it would reset the count before you'd be able to see the stats.

If you print from octoprint though we have the luxury to add additional gcodes that can help the printer. I proposed a while back to add M75/M77 for this purpose:

https://github.com/prusa3d/PrusaSlicer/issues/3724

M75 would be the perfect place for this IMHO. It's not implemented yet, but it's trivial to do.

Prusa-Support, 2022-07-05:
Thank you all for your feedback.

This was a quite specific and most interesting case. I seem to understand that culprit was most likely a failing Einsy board, but also the motor, the rods, and bearings on the axis were replaced.
Maybe the heat inside the enclosure added complexity to the case, similarly to issue #2653. You can check the issue if you are interested in the topic.

Is it safe to assume that the original issue does no longer persist?
Could this issue be closed?

Michele Moramarco
Prusa Research

scottmudge, 2022-07-06:
Yes I think the issue is corrected on my end, I will close. Thank you for the support.

closed 2022-07-06 by scottmudge