prusa3d/Prusa-Firmware #2941: [BUG] 3.9.3rc1 firmware decrease PINDA sensor relative to nozzle distance

jlandegent, 2021-01-02:
running latest 3.9.3 rc1 firmware on my MK3s (using no superpinda, think maybe V2?), noticed to succeed the xyz calibration test i had to decrease my pinda-sensor-to-bed distance by almost the same distance as my my nozzle distance otherwise the nozzle crashes into the magnetic bed. when using the older firmware i had not this issue. when calibrating the live-z values old +-1200 vs new +-1500 so that is correct, cause the distance pindasensor to bed is changed. the pindasensor needs to be too close to the bed for calibration using new firmware and when using the old tie-wrap distance, the distance is to great. now i used only the tip of the wrap and it will go ok without crashing the nozzle into the bed, but only just....

espr14, 2021-01-04:
Thank you for the report. We'll discuss possible solutions.

espr14, 2021-01-04:
As a hotfix you can lower the probe for the XYZ calibration only and then raise it for the printing. Or you can remove nozzle for the XYZ calibration.

jlandegent, 2021-01-04:
making the nozzle height almost the same as pinda height it will manage, but just slightly.....

3d-gussner, 2021-01-05:
@jlandegent I tried to test your results with my MK3S (with PINDv2) and I can **not** confirm your issue with the Live-Z value.
The xyz calibration ran longer with FW3.9.3-RC1 vs. FW3.9.2 but both Live-Z values have been very close to each other.
Can you provide some pictures of nozzle - PINDAv2 height difference before/after?
Which PINDAv2 colour do you have "grey" or "black" sensor tip?

Also some other things would be interesting to see using FW3.9.3-RC1 vs FW3.9.2 with identical PINDAv2 mount:

Connect your printer via USB and open Pronterface/Octoprint
- Run `G28W` ;Home without Mesh Bed Leveling
- Run `M114` ;Get current position
  - Should look like `X:12.00 Y:6.00 Z:0.15 E:0.00 Count X 12.12 Y:7.45 Z:0.15 E:0.00` where the Count values may differ depending on your xyz cal. 
- Check that the PINDAv2 LED is **off**
- Place x amount of paper sheets under the nozzle to "measure" the distance (write down how many sheets you need to touch the nozzle). If you have another way to measure the distance feel free to use it.
- Run `G91` ;Set to Relative Positioning
- Run `G1 Z0.01` ;Move Z by 0.01mm up _watch PINDAv2 LED_
- Run `M114` to see if it moved up 
- Re-run last two steps until PINDAv2 LED turns **on** and write down the Z value.

Report back your findings.
- Sheets / mm distance nozzle to steel sheet after `G28W`
- Z height when the PINDAv2 LED turns back on.
- Live-Z values FW3.9.3-RC1 and FW3.9.2

hlipka, 2021-01-08:
I see something similar - after upgrading from 3.9.2 to 3.9.3 I needed to change my Z distance from -0.980 to -0.680 (meaning my nozzle actually was hitting the bed at the beginning of doing the Z-distance calibration). I did not run the XYZ calibration after update - should I have done that? (the release notes say no)

CynanX, 2021-01-24:
Just to note, I upgraded to official 3.9.3, along with updating from the PRINDAv2 to the SuperPINDA and set the probe height with the the centre of the zip tie.

The probe is the same height as I had it before but my live Z has also changed from -1.200 to -1.670 range.

If the SuperPINDA is even a bit higher then the nozzle will hit the heat bed during calibration.

espr14, 2021-01-25:
If your live Z is lower then your nozzle is higher when the probe triggers. So there's extra 0.47 mm of clearance compared to PINDAv2 which is great.

Tahx, 2021-02-01:
Same issue.

My live Z was -2.03 and now it is -2.45, had to lower the SuperPINDA because otherwise the calibration would fail.

However the calibration is now more repeatable, been getting the same skew numbers over 3 or 4 XYZ calib.

Prusa-Support, 2022-05-14:
Hello! I am sorry for the late response to this thread. Is the issue still there with the newest firmware versions released in the meantime? @Tahx in case you still have issues with the newest stable FW, can you please let me know via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro")? We can review it in more detail from there. 

Alessandro Pantaleo
Prusa Research

closed 2023-11-02 by github-actions[bot]