prusa3d/Prusa-Firmware #2942: [BUG] Hard FW crash after check_file() returns false inside of menu_action_sdfile()

scottmudge, 2021-01-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S+
**Printer firmware version**-  3.9.2

**MMU Upgrade** - None

**SD card**

**Describe the bug**
If an incomplete gcode file is loaded, and the user clicks "No" during the "File incomplete. Continue anyway?" prompt, and then attempts to load a complete gcode file, the firmware performs a hard reset.

My OctoPrint serial log shows:

```
***<INCOMPLETE PROMPT HERE, USER INPUTS "NO">***
2021-01-02 20:58:56,020 - Recv: echo:busy: paused for user
2021-01-02 20:59:04,474 - Recv: echo:Now fresh file: 10X20_~2.GCO
2021-01-02 20:59:04,564 - Recv: File opened: 10X20_~2.GCO Size: 373774
2021-01-02 20:59:04,565 - Recv: File selected
2021-01-02 20:59:08,525 - Recv: start
2021-01-02 20:59:08,526 - Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
```
 
Unfortunately not very much debug output is provided.

**To Reproduce**
See above.
  
**Expected behavior**
No crash/reset.

leptun, 2021-01-03:
I fixed this bug recently and it got merged into the MK3 branch. No release firmware includes the fix yet, but it should come in 3.10.

Fixed by #2870

scottmudge, 2021-01-03:
Thanks, I will merge w/ current MK3 branch head for now.

closed 2021-01-03 by scottmudge