prusa3d/Prusa-Firmware #2943: [QUESTION] Proper print fan PWM configuration to remove/eliminate awful PWM noise?

scottmudge, 2021-01-03:
I've noticed that when using fan speeds below 80%, there is an awful low-frequency PWM noise. It is *very* audible. In fact, it is one of the loudest noises produced by the printer.

From looking at the source code, it looks like the print fan PWM control is being controlled in software, rather than hardware. And indeed, the PWM resolution is 4-bits, leading to a primary control frequency of ~62.5 Hz. Exactly the frequency that is very audible.

I would assume using software-based PWM control also imposes some overhead on the CPU as well. 

It's noted in `Configuration.h` that the software PWM "uses a very low frequency which is not ass [sic] annoying as with the hardware PWM."

Indeed, higher frequency noice is definitely more annoying, but I do know that there are methods of increasing PWM frequency beyond that easily heard by humans. 

It looks like this has already been implemented with the currently disabled `FAST_PWM_FAN` preprocessor directive.

Is there any downside to enabling `FAST_PWM_FAN` and disabling `FAN_SOFT_PWM`? It's noted that it may make the FET/Arduino run hotter, but is there any other risk or downside? 

I would be happy to add a heat-sink to the appropriate FET if it means eliminating the noise. Which FET is that which is responsible for the print fan?

scottmudge, 2021-01-03:
After some googling, it looks like FAST_PWM_FAN likely isn't a viable option due to issues with interference with measuring fan speed. 

Which leaves me with 3 options:

1. Increasing `FAN_SOFT_PWM_BITS` to 5, to reduce frequency down to 31.25 Hz. A little lower, maybe not as annoying? Perhaps it can't effectively control the fan at this resolution.
2. Adding a capacitor across the fan power pins. 
3. Adding a PWM-to-Analog converter module to convert the PWM signal to a variable DC voltage, from 0 to 5V.

DRracer, 2021-01-12:
please use and fill the predefined issue templates ... what machine are you talking about? Any mods?

scottmudge, 2021-01-12:
No mods, standard MK3S+.

The part/cooling fan shipped with the kit was not meant for PWM-based speed control, but instead was meant to be set at a static speed, or controlled via voltage modulation.

PWM *does work*, but the noise produced was significant.

The noise was significantly reduced by placing a 470-ohm capacitor across the positive and negative inputs to the fan, but this produced significant upstream noise in the MCU (probably doesn't have enough filtering on the I/O pins?), which led to constant firmware crashes when the fan was turned on or off.

Ultimately, I think it's simply a limitation of an "analog" fan being paired with a digital output. Reducing the effective range of the noise from the ~60 Hz range to 30 Hz range did help reduce the noise, but I think that was due to particular acoustics of the place where I put my printer.

closed 2021-01-12 by scottmudge