prusa3d/Prusa-Firmware #2950: [BUG]Translation mistake in spanish version on the messages

David-DL, 2021-01-07:
**Printer type** - MK3
**Printer firmware version**-  3.9.1-3518

**SD card or USB/Octoprint**
  Printing via Octoprint

**Describe the bug**
  The message that appears on the screen of the printer after a filament change is done has a misleading translation error in the spanish version.
The original english message says:
- "Was filament unload sucessful?"
And the translated spanish message says:
- "?Se cargocon exito el filamento?"

The spanish version will translate as: "Was the filament loaded sucessfully?" (asking the opposite of the original message).

_From lang/lang_en_es.txt:_
```
#MSG_UNLOAD_SUCCESSFUL c=20 r=2
"Was filament unload successful?"
"?Se cargocon exito el filamento?"
```
  
**To Reproduce**
  - Start a print with a filament change and wait for the filament change.
  - Press the knob to unload the filament when indicated.
  - The message will appear after the filament is unloaded

**Expected behavior**
  The following message should show up:
  - **¿Se descargo con exito el filamento?**
 
_(Ignoring accents as I imagine the screen has limitations on that)_

3d-gussner, 2021-01-08:
@David-DL Thanks for the detailed bug report, and the video.
You filled in the template with all information we ask for :medal_sports: 

I am not sure if you are familiar with github and creating a pull request, but if you are please consider to create a PR for this. Please don't forget that the LCD is limited what characters can be shown, so a '**¿**'  must be a '**?**'.

If you aren't please let us know and we will create one for you.

David-DL, 2021-01-08:
Hello @3d-gussner .
Sure, I'm familiar with it. Will send the pull request a little later today.
About the question marks, I think in that case it'd be better to just have the closing mark '?'.
Thanks!

gudnimg, 2023-07-04:
Looking at the translation today it seems this change/fix is already merged. So I'm closing this issue as resolved 👍 
[image]

closed 2023-07-04 by gudnimg