prusa3d/Prusa-Firmware #2952: [ENHANCEMENT] MINTEMP Tuning

rvalle, 2021-01-08:
**Printer type**

MK3S

**Is your enhancement related to a problem? Please describe.**

MINTEMP is not working in certain environments.

I have my printer in the store, which is normally around 10-13 C degrees. I monitor my prints via Octoprint.

I built a Lack enclosure to help printing. A simple preheat before printing will bring all temperatures into range. We even wrote a gcode that will ensure the ambient temperature is right before starting to print.

With the current MINTEMP setup the printer just refuses to start taking commands from octoprint, as it reports an error.
  
**Describe the solution you'd like**

It would be ideal to be able to configure the Minimal temperature, I would set it up to 10 degrees or whichever temperature that my environment supports. 

Alternatively It would be good to setup whether mintemp should be an error or just a warning on the display. 

Disabling the check is also a possibility but I read that this check can also detect cabling problems, for which just tuning the temperature would be the best solution.

psy0rz, 2021-01-11:
I have 3.9.1 and experience this problem as well. I think in earlier versions i could still heat the printer by manually selecting the temperature i think?

ra-ma, 2021-01-24:
Same problem here too. The connection to Octoprint is killed because the printer goes into an serious/watchdog error and has to be reset at the machine.

According to the code this is intended. Potential solution: Decrease min. temp and/or increase allowed time period for heating up.
https://github.com/prusa3d/Prusa-Firmware/blob/46c74f84b6c09b6cc6863476fae21f44d34b037c/Firmware/temperature.cpp#L2077

IntegersOfK, 2021-04-30:
Also somewhat related to #3053.

rvalle, 2023-10-26:
the issue is still ongoing, in winter printers used in a garage/storage room, require a external source of heat (i.e. hair dryer) to get started.

The easiest would be just to allow for this check to be disabled, just like many others are.