prusa3d/Prusa-Firmware #2959: Extruder setting may activate when hotend is not hot enough, potentially damaging the extruder/hotend

jasonbcox, 2021-01-12:
Shortly after completing a PETG print, I intended to raise the Z-axis to inspect the nozzle (via Settings -> Move Axis). I accidentally went a click too far and ended up selecting Extruder. I cranked the value up to 100 before I realized my error. During this time, the nozzle had cooled significantly, rendering the hotend unable to extrude at all. However, it did not stop despite the temperature dropping too low. It was trying to cram cold filament through the nozzle. I quickly searched for some cancel action, for which I could find none at either Settings or top-level menu. In order to prevent damage to the machine, I flipped the power switch.

To summarize two issues here:
1. Extruder option in the Move Axis menu should not activate unless proper temps are reached.
2. Cancel option does not exist when Extruder is active

scottmudge, 2021-01-12:
A similar event happens when an external extruder command is received over the serial line. This has happened to me, and the extruder gears slipped, causing loud clicking noises. This is effectively an artificial extruder jam. I believe the design of the extruder is to allow transient gear slips without damaging the extruder if something like this occurs.

Obviously if the extruder was constantly running with a cold hotend, it would end up wearing out the gears, but I don't think a transient episode like this will cause lasting damage.

I also believe the feature isn't temperature-gated, as some users may find it useful to use the extruder in the negative direction regardless of hotend temperature.

jasonbcox, 2021-01-12:
This is good information. Thanks. I didn't realize you could have the extruder go in the negative direction, but that makes sense.  If someone can confirm the extruder design regarding gear slippage, that would also be helpful.

neophyl, 2021-01-18:
One other use case here that people tend to forget is that some people modify their printer to print chocolate or other materials other than plastic.  Yeah I know :)  
In this use case the minimum printing temperature is considerably lower than that of most plastics if heated at all.  If the firmware is modified to not allow extrusion below a certain temperature then that would make using it for non plastic material not possible.

yoyo42, 2021-02-07:
It would make sense to have the ability to disable the 'don't move unless hot' check for those rare cases, but in general for most users an extruder lockout when cold would be of benefit. I've come across it while changing filament too, if I go away from the printer for a while I can get stuck in the middle of the change filament sequence with a cold hotend and have no choice but to grind the extruder or reset the printer. I'll post that as an issue later on, but what it shows is that it shouldn't jus be a block but instead if people try to extrude while cold the display should offer to heat it up to suitable temp then continue with the command.

closed 2023-11-02 by github-actions[bot]