prusa3d/Prusa-Firmware #2966: WTF Prusa

BigJBehr, 2021-01-13:
I would like to know why the MK3 Firmware (any version) has a MINTEMP_BED threshold of 10C when the MK2 firmware (all versions after 2.7.0 and all MK2 variants) have a MINTEMP_BED threshold set to 30C ?
The 30C setting causes my MK2.5S to constantly throw a MINTEMP Bed error. When I turn the printer on the ambient temperature of the hot end and print bed reads 22C (about 72F). This is below the MINTEMP_BED threshold, so I get the error.
Why was this change made ?
Would you please put it back to 10C and release a new version.
I have loaded version 2.7.0 firmware so that I can use my printer again without getting the error.
FIX IT.

leptun, 2021-01-14:
Please tell me, what temperature does the display say when the printer is really cold? Because if it is 30, you just got the answer to why it’s set to 30 in the firmware. The hardware is simply incapable of reading temperatures below 30. This is a flaw in the design of the motherboard thermistor circuitry that was addressed in the newer EinsyRambo board for the MK3 and such.
More in depth info here. https://github.com/johnnyr/Mini-Rambo/issues/8

DRracer, 2021-01-14:
@BigJBehr first of all - naming an issue like you did is unnecessarily rude.
Moreover - you've written the answer yourself: _When I turn the printer on the ambient temperature of the hot end and print bed reads 22°C_ ... while having 10°C in your room.

@leptun is correct, due to electrical wiring of the thermistor on the board the CPU is unable to detect lower temperatures reliably. Therefore the MINTEMP limit is set to a safe value for MK25/S which still can be read correctly from the thermistor (distinguishing from a disconnected/broken thermistor).

So the answer is **no**, we won't _FIX IT_ - there will be no change to this particular feature, however we are thinking about some manual override procedure, which would allow cold machines (starting below their operating limits) to be heated up while the user is closely watching them.

Closing this issue...

closed 2021-01-14 by DRracer

BigJBehr, 2021-01-14:
Yes it was rude, I wanted to get a quick reaction and it worked. However, I do not buy your explanation. When I revert back to firmware 2.7.0 I no longer have the issue. I do not consider this issue closed until I have an updated working version of the firmware for my MK2.5S.
Also, when looking for answers about why I was having this problem all I kept seeing from Prusa was that it was a bad or disconnected thermistor. No mention was made of a flaw in the mini-rambo circuitry. I spend more than a year living with a problem that I could not fix by following your advice. So after replacing the heated bed and a FET on the mini-rambo I found that it was a software issue.