prusa3d/Prusa-Firmware #2970: [FEATURE REQUEST] G4 "sleep..." message and M117

daBernd, 2021-01-14:
In custom scripts I often use the G4 command. When used alone there is no message on the printer display. But used as G4 P1000 or G4 S2 the display shows sleep... - which does not disappear after the G4 command.
Also if you want to display a message like: "M117 Waiting for 30 sec" this message is replaced by the "sleep..." when followed by a "G4 P30000" command.

My Request is to add a message string option to the G4 command, or to check wether there is something written by a previous M117 command in the display already, and then not to replace it with a not-anything-saying "sleep..."

Also, let the "sleep..." disapear when the pause is over.

Best regards...

buzkill420, 2022-01-09:
Up-vote!

gudnimg, 2022-04-17:
This can be fixed similarly as for M601 in https://github.com/prusa3d/Prusa-Firmware/pull/3475

PR #3475 (2022-07-04): Make M117 message persistent on LCD when used with M601 or G4
If M117 is called before M601 or G4 then we would like the custom message to
be visible on the LCD screen until the print is resumed.

Change in memory: +16 bytes of flash

Fixes #3316 (M601)
Fixes #1443 (G4)
Fixes #2970 (G4)

Video demonstration (with M601):

* Before (current Master branch): https://www.youtube.com/watch?v=8Mr-1lYwjsk
* After (this PR): https://www.youtube.com/watch?v=A7_uLBHg0qE
commit bf38241a (2022-07-04): Make M117 message persistent when using G4 command

This is also supported in Marlin 2 FW

Fixes #1443
Fixes #2970
closed 2022-07-04 by leptun