prusa3d/Prusa-Firmware #2971: [BUG]<xyz calibration fail with the new firmware on mk3s+>

MaddoxHU, 2021-01-15:
**Printer type** - [MK3S+]
**Printer firmware version**-  [3.9.3]

**Describe the bug**
I have a few weeks old MK3S+. With the 3.9.2 the XYZ calibration was always successful. I upgraded my firmware to the new 3.9.3. With that new XYZ calibration at the end of the calibration after the 9 points where I have to put a a paper between the nozzle and the heatbed I get that message: "xyz calibration failed. please consult the manual". I tried 2 times and both I got the same failed message. I changed back the firmware to the previous one and with that the calibration was successful at the first time...

snympi, 2021-01-25:
Is this related to [#2954](https://github.com/prusa3d/Prusa-Firmware/issues/2954)?

leptun, 2021-01-25:
Maybe, but I think it’s probably something related to the new XYZ calibration algorithm.

driftx, 2021-02-02:
I have this problem also, but on the MK3S.  I've gone back to 3.9.2, but did try 3.9.3 again after that to no avail, it always fails calibration.

N0ciple, 2021-02-04:
I had the same problem on a MK3S with the firmware 3.9.3. Changing mode from `stealth` to `normal` did the trick : the XYZ calibration did not failed with `normal` mode.

I hope this helps !

driftx, 2021-02-04:
I wish it did but I was always in normal mode.

closed 2023-11-02 by github-actions[bot]