prusa3d/Prusa-Firmware #2972: [FEATURE REQUEST] Streamline changing nozzle

amarilio, 2021-01-15:
Applies to all prusa FDM printers

**Is your feature request related to a problem? Please describe.**
Changing nozzles is a tedious and error prone process:
1. Preheat printer - if you change nozzles on a cold printer you could damage your printer
2. Replace nozzle using two tools - side note: there are now good tools in the market to help with that (e.g. Zcatch 1/4 drive nozzle removal tool), it'd be great if Prusa provided a similar tool in the box.
3. Dig inside printer settings to change the nozzle diameter - to provide you extra safety against incorrectly sliced code
4. Perform a first layer calibration - or you'll have an uncalibrated bed

I made every mistake here on the list. :(
  
**Describe the solution you'd like**
The main problem here is user education. 
One potential solution would be to add a new "Change Nozzle" option to the menu (ideally in a highly discoverable location). Selecting it will walk you through the following flow:
1. Notifies you that it is preheating the nozzle to the right temperature
2. Prompts you to select the new diameter (by default it has the current diameter highlighted as most changes maintain the diameter)
3. Wait for nozzle to reach the target temperature
4. Notifies you - "You can now replace the nozzle - Press the knob when done." 
5. Asks you whether you'd like to perform a new first layer calibration
6. The new first layer calibration should use extrusion width and layer height that matches the nozzle diameter (e.g. a 0.25mm nozzle should not use a 0.2mm layer height but 0.15mm or lower. Extrusion width should also be respectively adjusted)

Hooch180, 2021-01-16:
Love this idea

bencobley, 2021-02-13:
Great suggestion! 

I often forget to Calibrate Z after a nozzle change, sometimes causing the nozzle to crash into the bed. @amarilio's proposal would prevent that.

Prusa-Support, 2022-07-19:
Thank you for the input.

A "nozzle change wizard" sounds very promising and really makes sense in my opinion.

For now, a quick way to preheat the nozzle adequately before a nozzle change is preheating for PC. It would preheat to 275°C instead of 285°C but it should work just fine if the nozzle is clean enough (I personally recommend a cold pull before changing the nozzle). If you printed a high-temperature filament (like PC), then heating more than this could be necessary.

I honestly don't know how frequently "an average user" chages the nozzle, but this feature request may gain popularity among the community over time.

Michele Moramarco
Prusa Research

wavexx, 2022-12-12:
This should be fixed by the new "Nozzle change" menu #3774

Prusa-Support, 2023-03-07:
Thanks for the input.
It will be closed now.

Michele Moramarco
Prusa Research

closed 2023-03-07 by Prusa-Support