prusa3d/Prusa-Firmware #2974: [BUG]<Wireless Firmware upgrade to MK3 not working in Mac OS v10.14.6>

mc-ott, 2021-01-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Mk3 Printer connection wireless with  Octoprint (latest version).
Mk3 present Firmware v3.9.2

Using Prusa Slicer 2.3 and trying to update the firmware, a connection to Octoprint cannot be established. Mac OS is 10.14.6. When I go tohrough the process of setting the Octoprint printer connection, after I put the IP and the API, "Browse" button opens a "Network Lookup" window which shows "Searching for devices: Finished" but the List below is Empty. However, when I hit the "TEST" button next to Browse, it finds the printer and says "Connection to OctoPrint works correctly".
I can also upload a gcode file to Octoprint wirelessly from the Prusa Slicer 2.3.
[image]
[image]

Looks like Wireless Firmware Upgrade has a bug at least in Mac OS 10.14.6.

Thanks
  
**To Reproduce**
Try to make a wireless firmware upload from Mac OS 10.14.6 .
  
**Expected behavior**
 Firmware Flasher does not find the correct serial port.
[image]

awenelo, 2021-01-18:
For not being able to find the printer in the "browse" window: It looks like this is an issue with PrusaSlicer and does not involve the firmware for Original Prusa i3 printers. You should probably file an issue in their GitHub repository, [here](https://github.com/prusa3d/PrusaSlicer).

For not being able to do wireless firmware updates: As far as I know, PrusaSlicer doesn't support wireless firmware upgrades. If you're looking to do wireless firmware upgrades, you should look at a firmware updater plugin for Octoprint, such as [this one](https://plugins.octoprint.org/plugins/firmwareupdater).

P.S. You should also probably re-generate your API key for Octoprint.

mc-ott, 2021-01-18:
Thanks, I'll check the Octoprint firmware updater.

closed 2021-01-18 by mc-ott

3d-gussner, 2021-01-18:
Updating the firmware via Octoprint firmware updater may cause issues as it can't upload the multi language part to the Einsy board, so be careful with that.
If you have a MK2/S, MK2.5/S then updating the printer via Octoprint firmware updater shouldn't be an issue as the 2nd language is stored in the MEGA2560.

MK3/S/+ firmware updates should be always be done via USB and PrusaSlicer!

mc-ott, 2021-01-19:
@3d-gussner , I saw your comment just after updating my MK3 to fw 3.9.3 with Octoprint Firmware Update plug-in and did not have any issues during or after. I am using the latest Octoprint version, maybe they have sorted out the issues you mention. Thanks for the info anyway.

JohnOCFII, 2021-01-19:
> MK3/S/+ firmware updates should be always be done via USB and PrusaSlicer!

First I've heard of this concern.  I just did the 3.9.2 to 3.9.3 update over the weekend using OctoPrint Firmware updater plug-in (which just seems to call avrdude)  - seemed to go without issue.  Is this because I use the default English installation?