prusa3d/Prusa-Firmware #2975: [BUG]<Initial setup calibration time incorrect>

limited660, 2021-01-18:
**Printer firmware version**-  3.9.3

**Describe the bug**
  Display still shows 12 minutes for xyz calibration.
  
**Expected behavior**
  Update message to new time for the redesigned calibration of 3.9.3

https://github.com/prusa3d/Prusa-Firmware/blob/134f841380f8dfc704c28d9f0d598782bf732e81/lang/lang_en.txt#L365

3d-gussner, 2021-01-19:
Thanks for the comment, good point.

gudnimg, 2023-06-22:
Issue was fixed here https://github.com/prusa3d/Prusa-Firmware/commit/7fef570d1b5230cd705f5d5d8219849772022c6d recently in 3.13 so I'm adding this to the 3.13 Milestone as a reminder for us to close this issue :)

gudnimg, 2023-07-31:
Closing this as the fix is released with FW 3.13.0

closed 2023-07-31 by gudnimg