prusa3d/Prusa-Firmware #2978: [ENHANCEMENT] Allow backup and restore of some stored settings around a factory reset

tg73, 2021-01-19:
**Printer type** - All
**MMU Upgrade** - N/A

**Is your enhancement related to a problem? Please describe.**
I was recently asked by Prusa Support to do a factory reset as part of diagnosing/resolving a problem. This would mean losing the per-sheet live-z values for all the steel sheets I have set up. It would be a tedious and time-consuming task to recalibrate all the sheets. Luckily we were able to solve the problem without resorting to factory reset, but I would prefer to be able to do a factory reset when needed without a terrible penalty to get the printer set up again afterwards.
  
**Describe the solution you'd like**
I would like some mechanism for backing up and restoring settings like the per-sheet live-z. In the first instance, this could be via gcodes to be issued at a terminal, although integration with PrusaSlicer/flashing would be great - eg, "flash and factory reset", with checkboxes for which settings to retain. This feature should be available for all settings which are determined by user intervention (as opposed to by automatic self-calibration).

3d-gussner, 2021-01-20:
@tg73 Thanks for the idea.
Not sure if/when this will be implemented, but meanwhile have a look at https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/eeprom.h#L39-L329 where the EEPROM is documented.

You can use gcode `D3` to read and write the EEPROM, **please be very careful!**
See https://help.prusa3d.com/en/article/prusa-specific-g-codes_112173#d-codes and don't forget that the parameters `A C X` are case sensitive.

Example:
- To backup Sheet1 name use gcode `D3 Ax0d49 C7` with a result like this
```
Send: D3 Ax0d49 C7
Recv: D3 - Read/Write EEPROM
Recv: 0d49  53 6d 6f 6f 74 68 31
Recv: ok
```
`53 6d 6f 6f 74 68 31` hex to ASCII is `Smooth1`

- To restore Sheet1 name use gcode `D3 Ax0d49 X53 6d 6f 6f 74 68 31` with a result like this
```
Send: D3 Ax0d49 X53 6d 6f 6f 74 68 31
Recv: D3 - Read/Write EEPROM
Recv: 7 bytes written to EEPROM at address 0x0d49
Recv: 0d49  53 6d 6f 6f 74 68 31
Recv: ok
```
If you change for example the last hex value `31` to `32` you will see on LCD that the name changes to `Smooth2`

**As said please be very careful with manipulating the EEPROM, as this may do harm to your printer or prints and a factory reset may be needed.
Changing the EEPROM manually are at own risk.**

tg73, 2021-01-22:
For reference, here's how to save and restore all 8 of the sheet configurations. This is based on the EEPROM definitions at the time of writing, where `EEPROM_SHEETS_BASE` is at address `0x0d49`, there are 8 sheet configurations, each occupying 11 bytes. As per `struct Sheets`, the 89th byte of data contains the index of the active sheet.

**Please read and be aware of all the risks and warnings highlighted by @3d-gussner in this thread. Changing the EEPROM manually like this is at your own risk.**

Read all sheet blocks:
```
D3 Ax0d49 C89
```
```
Send: D3 Ax0d49 C89
Recv: D3 - Read/Write EEPROM
Recv: 0d49  53 6d 6f 6f 74 68 31 70 fe 3c 22 53 6d 6f 6f 74
Recv: 0d59  68 32 a9 fe 00 22 54 65 78 74 75 72 31 1a fe 00
Recv: 0d69  1a 54 65 78 74 75 72 32 ff ff ff ff 43 75 73 74
Recv: 0d79  6f 6d 31 ff ff ff ff 43 75 73 74 6f 6d 32 ff ff
Recv: 0d89  ff ff 43 75 73 74 6f 6d 33 ff ff ff ff 43 75 73
Recv: 0d99  74 6f 6d 34 ff ff ff ff 00
Recv: ok
```

To restore:

Regex search and replace the lines starting `Recv: 0d`:
Search: `Recv:\s+([0-9a-f][0-9a-f][0-9a-f][0-9a-f])\s+(.*)`
Replace: `D3 Ax\1 X\2`

which gives:

```
D3 Ax0d49 X53 6d 6f 6f 74 68 31 70 fe 3c 22 53 6d 6f 6f 74
D3 Ax0d59 X68 32 a9 fe 00 22 54 65 78 74 75 72 31 1a fe 00
D3 Ax0d69 X1a 54 65 78 74 75 72 32 ff ff ff ff 43 75 73 74
D3 Ax0d79 X6f 6d 31 ff ff ff ff 43 75 73 74 6f 6d 32 ff ff
D3 Ax0d89 Xff ff 43 75 73 74 6f 6d 33 ff ff ff ff 43 75 73
D3 Ax0d99 X74 6f 6d 34 ff ff ff ff 00
```

Send each of the `D3` lines, one by one, to restore.

3d-gussner, 2023-10-26:
This has been solved with introduction of `M850` https://github.com/prusa3d/Prusa-Firmware/pull/2201

closed 2023-10-26 by 3d-gussner