prusa3d/Prusa-Firmware #2979: [BUG] "Extra Loading Distance" Setting for MK3s + MMU2s

carterjb, 2021-01-21:
Printer type: MK3S
Printer firmware version: 3.9.2  (Mosquito 450C Thermistor Table Added)
Customizations: Mosquito Hotend and Thermistor

MMU Upgrade: MMU2S
MMU upgrade firmware version: 1.0.6

SD card or USB/Octoprint: SD Card

**Describe the bug**
I am having an issue with the "Extra Loading Distance" feature within PrusaSlicer for my MK3s + MMU2s printer (Printer Settings -> Single extruder MM setup -> Extra Loading Distance). I am unsure if this issue is resolved in the firmware or in PrusaSlicer (Please inform, PrusaSlicer-Settings issue open as well: https://github.com/prusa3d/PrusaSlicer-settings/issues/114). I have the Mosquito hotend and thermistor upgrades, that is the only difference for my printer. I think my customizations may have possibly changed the distance between the extruder gears and nozzle slightly. I do not believe this is the root cause of my issue with this setting, however, please read on.
 
I believe there is an issue when loading after the filament reaches the IR sensor, it is driven slightly too far into the hotend leading to a blob every filament change. Please note, that I have tuned my filament changes to work quite perfectly other than this (Cooling moves, sensor calibration, etc.). These blobs are large enough to cause nozzle collisions and skipped steps within the wipe tower. Reducing "Filament Parking Position" seems to solve this issue, but causes another issue because upon unloading the filament isn't driven far enough.

There appears to be a feature in place for this exact problem, called "Extra Loading Distance", which can be negative. This would be a perfect setting for my problem, however, making the value negative doesn't seem to shorten the initial distance the filament is driven. Instead, I am observing it drive the filament forward by "Filament Parking Distance" first, then adjust by "Extra Loading Distance". However, this sequential retraction for negative values of "Extra Loading Distance" is not resolving my issue. Essentially, I would like it to adjust the true initial loading distance, such that the remaining filament in the hotend isn't accidently extruded creating this blobbing issue.

For example, I have even tried "Filament Parking Distance" = 85mm and "Extra Loading Distance" = -75mm as an extreme case. I observed it load the filament 85mm creating a blob, then retract 75mm such that the hotend didn't extrude anything for a long time during the wipe and print (obviously there was a void in the hotend). This is definitely not how I would need this setting to work in order to resolve my problem. In this extreme case, I would like it to instead just drive filament by 10mm toward the nozzle, which shouldn't create a blob. Then, I could tweak this setting such that remaining filament in the hotend won't cause these blobs while loading.

**To Reproduce**
Happens at every filament changes under stock profile slicer settings.
  
**Expected behavior**
Creates blobs due to loading the filament into the nozzle, likely the result of residual filament in the hotend. These blobs make the printer unusable as they cause collisions with the nozzle.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**

[image]

rickm0804, 2021-01-26:
I'm also having this issue or something similar when trying to print with the MMU2. I'm using the latest firmware 3.9.2 and everything is stock on my MK3S. Attaching pictures to hopefully help confirm if this is the same problem as carterjb. Any recommendations to help 
[image]
[image]

fix what's going on would be greatly appreciated! Thanks!
[image]

carterjb, 2021-01-28:
UPDATE: I performed some more testing yesterday, I confirmed that properly calibrating the IR sensor and updating PrusaSlicer did not resolve the issue. I have been testing with this simple multiple material print. I sliced my test model as usual using the stock printer settings ("PLA-PLA_Stock.gcode").

[image]

Furthermore, I isolated the issue in the GCode by moving the extruder off the purge/wipe tower in the Y-direction right before the blob was created (creating the blob outside it, added line "G1 Y140.00" previous to this section). I found during the filament loading portion that these lines specifically created the blob, as the blob was created at Y=140mm instead of on the wipe tower, then the extruder moved back to the tower. The modified GCode is called "PLA-PLA_Stock_DEBUG.gcode". I kept commenting the "G1 Y109.962" moves following my edit till I found that the printer had created the blob outside the wipe tower. It seems these lines following the M900 commands are what specifically caused the blob.

<Line 385>   G4 S0
; CP TOOLCHANGE LOAD
G1 E12.0000 F600
G1  X196.341  E42.0000 F552
G1  X164.157  E6.0000 F386
M73 P8 R26
M73 Q8 S26
G1  Y109.962 

It seems that I can resolve this issue possibly by post-processing the .gcode files with python and moving the extruder off the wipe tower prior to creating a blob. However this is obviously not an ideal situation. 

**Extra Loading Distance Setting:**

After setting "Extra Loading Distance" to -75mm in PrusaSlicer (as discussed in my original post), I can see that the GCode for this section looks different ("PLA-PLA_Extreme_Example.gcode"). However, the problem is not resolved at all and the blob is created of similar size. The only change in behavior is there is a massive retraction move right after the blob is created. This retraction move simply leaves a void in the nozzle and no filament is extruded for a long time. Interestingly enough, I don't even see this massive retraction move in the GCode in this section. In fact it looks as if the extrusion moves are reduced, so just by looking at this GCode I would assume the issue would be resolved but it is not. The printer behaves almost identically, except an additional problem is created from the retraction such that there is a void in the nozzle.

<Line 386> G4 S0
; CP TOOLCHANGE LOAD
G1 E2.0000 F600
G1  X193.324  E7.0000 F3000
G1  X164.157  E1.0000 F2100
G1  Y109.962 

**End Goal**

My end goal is simply to try and prevent this blob issue in the first place. I know the issue can be resolved by reducing the distance the filament is loaded into the nozzle, but I am unaware of how to do that properly. There is evidently code somewhere in the firmware, driver, or PrusaSlicer which is driving the filament to the nozzle while it is stationary (after triggering the IR sensor). The only fix I have found for the blobs is reducing the "Filament Parking Distance" in PrusaSlicer. However this makes the printer unable to unload the filament (gets stuck between the Bondtech gears), so it is hardly a workable solution. I will be working to post process my .gcode files with Python as a temporary fix for now, but I hope that this is resolved in the future.
[Testing_GCode_Files.zip](https://github.com/prusa3d/Prusa-Firmware/files/5889274/Testing_GCode_Files.zip)

closed 2023-11-06 by github-actions[bot]