prusa3d/Prusa-Firmware #2985: [FEATURE REQUEST] when selecting unload filament, do not pre heat the bed and option to toggle this functionality

oderbang, 2021-01-22:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** -  MK3

**Is your feature request related to a problem? Please describe.**
When selecting "unload filament" when the bed is cold the printer pre-heats the build plate. this is un necessary if you are just removing filament.
  
**Describe the solution you'd like**
have a toggle option in the setting to enable/disable build plate pre-heat when using filament unload.

snympi, 2021-01-25:
I would find this useful as well. Sometimes after completing a print my printer remains on, but idle and both the bed and nozzle cools down. When I want to change the filament in prep for a new print, just heating the nozzle would be much faster because when I print 10-30 minutes later the bed would have warmed up unnecessarily.

Panayiotis-git, 2021-01-25:
Please find an ongoing discussion on this subject [here](https://github.com/prusa3d/Prusa-Firmware/pull/2335)

Prusa-Support, 2022-05-19:
merge with #422

closed 2022-05-19 by Prusa-Support