prusa3d/Prusa-Firmware #2989: [BUG] File select from SD fails.

CyberHugger, 2021-01-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** -  MK3S+
**Printer firmware version**-  3.9.3

**MMU Upgrade** - None
**MMU upgrade firmware version n/a

**SD card or USB/Octoprint**
  Printing from SD card

**Describe the bug**
  Just finish building MK3S+. First print was a Marvin from the included SD card and that printed fine. Upgraded to 3.9.3 which finished successfully. Selected Print from SD and got the list of files. Tried to scroll down but the cursor disappeared and a few seconds later the screen went blank for a few seconds followed by the Prusa splash screen. That’s where it ended up stuck. There was no response to the control. I could press the Reset button to get it to reboot but then I got the same failure when attempting to print from the SD card. Also tried a power cycle but that did not help. Did try reflashing the firmware but that did not help. Flashing with 3.9.2 fixed my issue so I’m pretty sure it’s not a hardware issue or an SD card problem.
  
**To Reproduce**
  Flash the printer with 3.9.3 then try to print from the SD card. Scroll down to select a file and the cursor disappears.  
  
**Expected behavior**
  When printing from the SD card I should be able to scroll down and select a file to print.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2021-01-23:
Oh no. This could be related to #2954

CyberHugger, 2021-01-23:
The symptoms are different. That report says that they cannot get a successful boot after upgrading to 3.9.3. My printer boots up normally but then fails when I try to select a file to print on the SD card.

leptun, 2021-01-23:
I know the symptoms are different, but the thing is that nothing changed in that part of the code between 3.9.2 and 3.9.3. There still is no idea as to why 3.9.3 has random issues all over the place, so even something as unrelated as this could be caused by the mystery issue.
There have been reports of garbled menus in 3.9.3, so having a crash during the SD menu isn't that far fetched considering that the SD menu is notorious for being weirdly implemented.

DRracer, 2021-03-19:
@CyberHugger could you please verify your issue on the latest firmware 3.10-RC1 and report back?

JohnnyDeer, 2021-08-19:
I'm closing this issue because there are no responses

closed 2021-08-19 by JohnnyDeer