prusa3d/Prusa-Firmware #2994: [FEATURE REQUEST] Sort by time when inserting SD card and alphabetically when triggering print via menu

fe60, 2021-01-25:
**Printer type** - MK3S

**Is your feature request related to a problem? Please describe.**
Having created a new print job, I really like the jobs are ordered by time, when the SD card is inserted.
When triggering a job via menu (Print from SD), I prefer the list of jobs to be shown in alphabetical order. To get the right order for each situation, I have to change the ordering in the settings. That's annoying.
  
**Describe the solution you'd like**
Allow different ordering for "Print from SD" (alphabetically) and when inserting the SD card (by time). I don't know how to name such a setting. So far there are three options: Alphabet, None, By Time.
A new one could be named "Mixed" or "Alph./Time".

Kachidoki2807, 2021-01-25:
That's a nice idea. I would even go further.
When inserting the SD card, instead to have to scroll down (annoying if there are lot of directories), the latest G-Code in date could be pre-selected automatically.

wavexx, 2021-01-26:
My 2c: when going through the "Print from SD" menu I almost always the most recent prints.

However I agree that when sorting by time, directories should be sorted as well (mixed list of file/directory), or pushed behind files, so that the first file (preselected) would always be first.

closed 2023-11-06 by github-actions[bot]