prusa3d/Prusa-Firmware #2998: Change "FILAMENTCHANGE_XPOS" to X=0 or X=100 or make to choose or ...

nisrh, 2021-01-27:
Situation:
It is about filament change position. The default position "rightmost" on X-axis is suboptimal, because in case, something goes wrong with automatical filament ejecting at that Position and  i have to remove the old filament manually, there is to less space to catch the old filemant with the pliers or something else.
So this is a reason, why i loose a complete print with Color change, when something goes wrong with the Filament ejection. Can you please find a solution for that?

3d-gussner, 2021-01-27:
@nisrh Thanks for the Feature request but it is missing all the information asked in the template which should be filled in to avoid unnecessary questions. (If you like please close this one and create a new using the template).

Changing the position is a tricky one as some want these here other there. Please search the issues for similar issues, as I think there have been already one open. Discussions in one issue may have better chance if the community and Prusa can agree on one solution that fits the most.

nisrh, 2021-01-28:
@3D-gussner 

Thanks for the comment. 
Only thing i could find, i missed, what was asked for in opening an issue is the printer Type. 
Sorry, i did not mention that: It is a MK3S.

I described the problem in detail and left the solution open, because i mentioned it in the title of the issue.

Is there an other template, i missed? 

I searched for similar issues before and could not find one. Do you have a link for me?

closed 2023-11-06 by github-actions[bot]