prusa3d/Prusa-Firmware #2999: Idle Temp Shows 265C When SET Temp is at 0?

krkpatrck, 2021-01-28:
After 8+ hours of trying to get to this point im not even sure if ive done it correctly. however, i do have the firmware.ino.hex file flashed onto my MK3s. when i turn on the printer my temp reading shows 265/0. not sure what to do from here?

Panayiotis-git, 2021-01-28:
You have to copy the file "1_75mm_MK3S-EINSy10a-E3Dv6full.h" from the "variants" folder to the "firmware" folder and rename it to "Configuration_prusa.h". Then rebuild.

3d-gussner, 2021-01-28:
@krkpatrck Please try the `PF-build.sh` script and use the last stable source code https://github.com/prusa3d/Prusa-Firmware/tree/v3.9.3 as the MK3 branch is constantly on development.
`PF-build.sh` will download and configure a know/supported/stable Prusa Firmware build environment and guide you the process.

krkpatrck, 2021-01-28:
@3d-gussner  tried the PF-buld.sh scrpt and got the following.

dads-pc@Dads-PC ~]$ sudo /home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3/PF-build.sh 
[sudo] password for dads-pc: 
Linux 64-bit found

Script path : /home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3
OS          :
OS type     : linux

Arduino IDE : 1.8.5
Build env   : 1.0.6
Board       : prusa_einsy_rambo
Package name: PrusaResearch
Board v.    : 1.0.3
Specific Lib: PrusaLibrary

 1) Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h
 2) Firmware/variants/1_75mm_MK2-RAMBo10a-E3Dv6full.h
 3) Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h
 4) Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
 5) Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h
 6) Firmware/variants/1_75mm_MK25-RAMBo10a-E3Dv6full.h
 7) Firmware/variants/1_75mm_MK25-RAMBo13a-E3Dv6full.h
 8) Firmware/variants/1_75mm_MK25S-RAMBo10a-E3Dv6full.h
 9) All
10) Quit
Select a variant: 5

Which lang-build do you want?
1) Multi languages
2) English only
Select a language: 1
/home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3/PF-build.sh: line 638: cd: ../PF-build-env-1.0.6/1.8.5-1.0.3-linux-64: No such file or directory
[dads-pc@Dads-PC ~]$ 

Im pretty sure  have the most recent files i just downloaded them yesterday but i will check your provided link to be sure.

krkpatrck, 2021-01-28:
@Panayiotis-git i didnt try that. i dont think that was in the instructions if i remember. ill give it a shot though, after 8 hours of trying why not.

krkpatrck, 2021-01-28:
@3d-gussner i downloaded the new files with the link you provided. i get the same error i think.

dads-pc@Dads-PC ~]$ sudo /home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3/PF-build.sh 
[sudo] password for dads-pc: 
Linux 64-bit found

Script path : /home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3
OS          :
OS type     : linux

Arduino IDE : 1.8.5
Build env   : 1.0.6
Board       : prusa_einsy_rambo
Package name: PrusaResearch
Board v.    : 1.0.3
Specific Lib: PrusaLibrary

 1) Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h
 2) Firmware/variants/1_75mm_MK2-RAMBo10a-E3Dv6full.h
 3) Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h
 4) Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
 5) Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h
 6) Firmware/variants/1_75mm_MK25-RAMBo10a-E3Dv6full.h
 7) Firmware/variants/1_75mm_MK25-RAMBo13a-E3Dv6full.h
 8) Firmware/variants/1_75mm_MK25S-RAMBo10a-E3Dv6full.h
 9) All
10) Quit
Select a variant: 5

Which lang-build do you want?
1) Multi languages
2) English only
Select a language: 1
/home/dads-pc/Downloads/Prusa_Bear_Upgrade/build-env/Prusa-Firmware-MK3/PF-build.sh: line 638: cd: ../PF-build-env-1.0.6/1.8.5-1.0.3-linux-64: No such file or directory

3d-gussner, 2021-01-29:
@krkpatrck No need to use `sudo` to run the script. Also change to the folder before running the script, see https://github.com/prusa3d/Prusa-Firmware#compile-prusa-firmware-with-ubuntudebian-linux-subsystem-installed

krkpatrck, 2021-02-06:
@3d-gussner here is my current read out. been budy to look into this but i got this code today. 
1. i copied and changed the varient file to cinfiguration.h into the firmware folder.
2. i ran the build.sh script and NOT SURE WHAT TO DO NEXT. i have a build-env folder containing another folder i unzipped (PF-build-env-Linux64-1.0.6.1)
3. i ran the pf-build.sh script and had to run it originally as sudo because the folder it needed didnt exist so sudo created it. i could run it then without sudo if i wanted to retry. this is the code i now get if i run pf-build.sh

/home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/PF-build.sh Linux 64-bit found

Script path : /home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3
OS          :
OS type     : linux

Ardunio IDE : 1.8.5
Build env   : 1.0.6
Board       : prusa_einsy_rambo
Package name: PrusaResearch
Board v.    : 1.0.3
Specific Lib: PrusaLibrary

Unzipping Prusa Firmware build environment...
/home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/PF-build.sh: line 389: unzip: command not found

since the instruction are not at all clear at least to my uneducated self my 2 request are.
1. after running the build.sh script what do i do now with the files i unzipped, the next steps to take?
2. if you suggest how to fix the pf-build.sh script what will my next steps be. im on arch linux manjaro but you suggest the "Using Linux subsystem under Windows 10 64-bit" so this does confuse me some as to why.
3. thank you both for getting me this far.

3d-gussner, 2021-02-06:
Read the lines 12—26 https://github.com/prusa3d/Prusa-Firmware/blob/MK3/PF-build.sh

Normally when you start the script with `./PF-build.sh` and it is missing something it gives you a hit what and how to install it.
I have tested the script with Ubuntu and Debian maybe you need to lookup how to get `unzip` working in your Linux.

krkpatrck, 2021-02-06:
Thanks for the tip about unzip and scripts ill look into it

krkpatrck, 2021-02-06:
ok so i cant get pf-build to work. i installed the unzip dependicies i needed as advised by the manjaro community and now cant get past this error. 

```
/home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/PF-build.sh    Linux 64-bit found

Script path : /home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3
OS          :
OS type     : linux

Ardunio IDE : 1.8.5
Build env   : 1.0.6
Board       : prusa_einsy_rambo
Package name: PrusaResearch
Board v.    : 1.0.3
Specific Lib: PrusaLibrary

 1) Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h
 2) Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
 3) Firmware/variants/1_75mm_MK2-RAMBo10a-E3Dv6full.h
 4) Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h
 5) Firmware/variants/1_75mm_MK25-RAMBo10a-E3Dv6full.h
 6) Firmware/variants/1_75mm_MK25S-RAMBo10a-E3Dv6full.h
 7) Firmware/variants/1_75mm_MK25-RAMBo13a-E3Dv6full.h
 8) Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h
 9) All
10) Quit
Select a variant: 8

Which lang-build do you want?
1) Multi languages
2) English only
Select a language: 2
/home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/PF-build.sh: line 510: cd: ../PF-build-env-1.0.6/1.8.5-1.0.3-linux-64: No such file or directory
```
Yes they shouwed me how to contain the code to be easier to read, sorry about previous post not having it.

when i try the build.sh script i get this

```
/home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/build.sh 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Build options changed, rebuilding all
open sketch/Firmware.ino.cpp: permission denied
```
OR

```
sudo /home/dad/Documents/PRUSA/Firmware-pt100/Prusa-Firmware-3.9.3/build.sh 
[sudo] password for dad: 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 233650 bytes (92%) of program storage space. Maximum is 253952 bytes.
Global variables use 6430 bytes of dynamic memory.
lang-build.sh started
lang=en
count=368
offs=752
size=10251
lang=cz
count=368
offs=752
size=9814
lang=de
[E]: Text "Sind linke+rechte Z- Schlitten ganz oben?" is longer then definiton on line 73
count=368
offs=752
size=11302
lang=es
count=368
offs=752
size=11002
lang=fr
[E]: Text "Z~carriages gauche + droite tout en haut?" is longer then definiton on line 73
count=368
offs=752
size=11506
lang=it
count=368
offs=752
size=11121
lang=pl
count=368
offs=752
size=10386
lang-build.sh finished with success
config.sh started
 Arduino main folder: OK
 Arduino builder: OK
 AVR gcc tools:
   objcopy OK
   objdump OK
 Output folder: NG!
 Objects folder: NG!
 Generated elf file: NG!
 Generated hex file: NG!
config.sh finished with errors!
Config NG!

this is the current brick wall im bashing against. not pun intended
```

Panayiotis-git, 2021-02-06:
Sorry, I cannot help you with the build.sh script as I've never used it so far.

One thing I've noticed at your first post:
the file Firmware/configuration_prusa.h contains the lines:
```
// Uncomment the below for the E3D PT100 temperature sensor (with or without PT100 Amplifier)
//#define E3D_PT100_EXTRUDER_WITH_AMP
//#define E3D_PT100_EXTRUDER_NO_AMP
//#define E3D_PT100_BED_WITH_AMP
//#define E3D_PT100_BED_NO_AMP
```

If you want to define the E3D_PT100_EXTRUDER_WITH_AMP, then you should uncomment the first define like this:
```
#define E3D_PT100_EXTRUDER_WITH_AMP
//#define E3D_PT100_EXTRUDER_NO_AMP
//#define E3D_PT100_BED_WITH_AMP
//#define E3D_PT100_BED_NO_AMP

```

krkpatrck, 2021-02-07:
oh ok, ill do that instead. thank you

closed 2023-11-06 by github-actions[bot]