prusa3d/Prusa-Firmware #3008: [BUG] Cold extrude not saved during Power Panic

wavexx, 2021-02-01:
**Printer type** - MK2.5+
**Printer firmware version** - 3.9+
**MMU Upgrade** - N/A
**SD card or USB/Octoprint** - N/A

**Describe the bug**
A print started using the cold extrude feature (M302) will fail to resume printing after a power-panic, since the limit temperature is not saved.
  
**To Reproduce**
Start a print with cold-extrusion (any temp below 175):

```
M302 S150
M104 S150
```

Trigger PP. As soon as extrusion is started the printer will stop due to cold extrusion prevention.

**Expected behavior**
Printer should continue printing as usual.

This is unlikely to cause significant issues, since the current cold-extrusion limit is pretty low.

3d-gussner, 2023-11-23:
@gudnimg Can you please add M302 value to PowerPanic

3d-gussner, 2024-04-09:
Fixed with #4656

closed 2024-04-09 by 3d-gussner