prusa3d/Prusa-Firmware #3019: [BUG] Stopping print while heating up leaves printer in strange state, reset required to get it working again

fe60, 2021-02-07:
**Printer type** MK3S
**Printer firmware version**  3.9.3

SD card ~~or USB/Octoprint~~

**Describe the bug**
Stopping a print while the printer is still heating up, leaves the printer in a strange state.
I have to reset the printer to trigger a new print.
  
**To Reproduce**
  * Start a print from SD card
	Use an arbitrary existing .gcode file. You will stop the print job before it starts printing.
  * Wait a moment to see temperature of nozzle raising.
  * Stop the print.
  * Fine
    * LED of print bed indicating heating is switched off
    * temperature of bed and nozzle drops
  * Wrong
    * Display still shows "Bed heating..."
    * It's not possible to start a new print. I have to reset the printer to start the next print.
  
**Expected behavior**
  Triggering a new print shall be possible without a reset. Just as it works when a print is stopped while it is already printing.

scottmudge, 2021-02-12:
Did you wait until it reached the temperature specified in the print after cancelling?

It will resume normal behavior after reaching temperature and completing the last few buffered commands.

Cancelling simply stops reading from the SD card, but the printer still has to process whatever command it is actively working on. The heat-up command can't be cancelled once it's started. So either reset it by using the reset button (hardware interrupt), or wait for it to finish.

leptun, 2021-02-12:
@scottmudge Stopping an SD print during heatup should have worked correctly, so it is a but. Just octoprint can't be canceled unless you use some plugin which does blocking behaviour on the host side, not the printer side.

scottmudge, 2021-02-12:
The more you know!

sberkovitz, 2021-02-28:
I have observed this behavior a few times but I do not need to restart the printer. It seems to clear on its own in a minute or so (haven't timed it, but can if that's useful). 

In this state, if you press and hold the button, you will get Live Z Adjust versus setting the Z axis value.

fe60, 2023-11-01:
There was an older issue #2862, but it has been closed by a bot.

fe60, 2024-01-03:
The bot asked for a comment, it gets a comment.

gudnimg, 2024-01-03:
Assigned this to myself so it won't be closed. I can try test this when I find time. I think this issue is solved in 3.13.2... but I will have to test it to be sure.

fe60, 2024-04-16:
Works as expected in 3.13.3

closed 2024-04-16 by fe60