prusa3d/Prusa-Firmware #3028: [BUG] Long filenames are truncated in selection for printing

GaryAitken, 2021-02-13:
**Printer type** - MK3S
**Printer firmware version**-  3.9.3

**SD card or USB/Octoprint**  SD card

**Describe the bug**
Long filenames are truncated in the display for selection to print from SD card.  For example, the name
GMF_Sheet_DripTipWGroove_2_4x6_0.4n_0.2mm_PLA_5h23m.gcode
displays as
GMF_Sheet_DripTipWGroove_2_4x6_0.4n_0.2mm_PLA_5h23m.
Longer file names have even more truncated, so that at times it is impossible to determine which one is which if the first part is the same.
  
**To Reproduce**
Slice a file with a long name, write it to the SD card, and select it for printing.  The name displayed for selection will be truncated, as will the name being displayed while printing.
  
**Expected behavior**
The display already scrolls; but it should scroll the complete file name.
  
**G-code**
The error is not in the g-code, it is in the display software.

**Video**
I doubt if it's really needed in this case.

leptun, 2021-02-13:
The fatFS implementation used in the firmware is limited to 52 characters of long filename. There aren’t enough resources to increase this limit.

GaryAitken, 2021-02-13:
That's unfortunate.  You don't need longer vFAT entries; you only need to store the full name of the one file being scrolled, which you can read from the SD card when the selection changes.  If the vFAT entries are going to be shorter than the full 255 characters allowed, you don't need 52 character vFAT entries -- just enough to fit on the display, which is 20 characters.  Use the saved storage to read from the card when the cursor selection changes.

closed 2023-11-08 by github-actions[bot]