prusa3d/Prusa-Firmware #3031: [BUG]<Downgrade from 4.3.1 to 4.2. Is not possible

Af5211, 2021-02-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2021-02-13:
This is Prusa Mini related. Please post the issue here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues
Also, please fill the template next time.

JohnnyDeer, 2021-02-16:
This issue is solved in MINI repository, I´m closing.

closed 2021-02-16 by JohnnyDeer