prusa3d/Prusa-Firmware #3037: [BUG] M73 does not show any remaining time

fuso2001, 2021-02-19:
**Printer type** - MK3S+
**Printer firmware version**- 3.9.3 commit 3556

**MMU Upgrade** - MMU2S

**SD card or USB/Octoprint**
  Print via USB/Octoprint

**Describe the bug**
  M73 Commando only shows the completed percentages in the display. The remaining time display remains at 0.
  
**To Reproduce**
  Start a print and watch the display.
  
**G-code**
  G-Code was generated with PrusaSlicer and contains M73 commands

3d-gussner, 2021-02-20:
@fuso2001 Thanks for reporting. Please add next time that you compiled the firmware in the **Printer firmware version** part.

fuso2001, 2021-02-22:
> **Printer type** - MK3S+
> **Printer firmware version**- 3.9.3 commit 3556
> 
> **MMU Upgrade** - MMU2S
> 
> **SD card or USB/Octoprint**
> Print via USB/Octoprint
> 
> **Describe the bug**
> M73 Commando only shows the completed percentages in the display. The remaining time display remains at 0. The messages via the USB interface towards the host are OK and contain the new "C" object. Time to change is a "-1" value.
> 
> **To Reproduce**
> Start a print and watch the display.
> 
> **G-code**
> G-Code was generated with PrusaSlicer and contains M73 commands

> @fuso2001 Thanks for reporting. Please add next time that you compiled the firmware in the **Printer firmware version** part.

> Okay, thank you for the answer. I now set starttime to _millis () in the M862.3 command. It works now.

3d-gussner, 2021-03-18:
Should be fixed with https://github.com/prusa3d/Prusa-Firmware/pull/3067

JohnnyDeer, 2021-07-19:
I'm closing this issue as solved by mentioned #3067.

closed 2021-07-19 by JohnnyDeer