prusa3d/Prusa-Firmware #3041: [BUG] Pinda Probe seems not to work with 3.9.3 on MK3 and Pinda V2

mettworks, 2021-02-24:
**Printer type** - MK3
**Printer firmware version**-  3.9.3

**MMU Upgrade** - no

**SD card or USB/Octoprint** - Octoprint

**Describe the bug**

I tried to print with new FW 3.9.3 and deleted old settings complete, I used fresh Slic3r profiles too. I could after adjusting Z in the middle only print in the middle. Not left, right and so on.I should have Pinda V2 (grey). After downgrading the FW to 3.9.1 again and reset the whole printer, I could print on the whole bed again. 

I'm not sure if it's a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2941 ? It seems that the Pinda Measurement is completely ignored or wrong interpreted. 

Actually I'm back on old FW and could print after many hours/days of testing and searching. 
  
**To Reproduce**
Full Reset with 3.9.3, try to print on the sides.

JohnnyDeer, 2021-03-23:
Hi @mettworks, thank you for reporting. Does this problem only occur via OctoPrint or also when printing from SD? Does first layer calibration work properly?

JohnnyDeer, 2021-08-19:
I'm closing this issue because there are no responses.

closed 2021-08-19 by JohnnyDeer