prusa3d/Prusa-Firmware #3047: [FEATURE REQUEST] Ability to invert function of Filament sensor

Prutsium, 2021-03-02:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK3S
**MMU Upgrade** - MMU2S

**Is your feature request related to a problem? Please describe.**
More a created problem: Function of filament sensor required to be reversed
  
**Describe the solution you'd like**
At the moment i am working on getting an Orbiter extruder working on a Prusa what is in general not a big deal but since i also want it with the filament sensor as the idler door i run into i minor issue:

I would need the logic of the IR sensor to be inverted and couldnt find in the firmware a place to do it as there is alot of info that reverts back to specific Marlin parts.

To give you an idea whats happening:
[image]

As you see when the filament is out the IR sensor would now be a 1 and when its in a 0.

I would appreciate if you can guide me to the firmware part to change the logic there.

Many Many thans in advance.

Jerome

@3d-gussner Thanks for the reply on my email.

Prutsium, 2021-03-02:
For the mean time a made the lever a bit different so no inverting would be needed but this would work less reliable.
https://www.youtube.com/watch?v=55Nbrctt00Q

3d-gussner, 2021-03-02:
Please search for IR_SENSOR_PIN and fsensor in the source code. You should find quite a lot.
There is no INVERT for this pin
With FS v.0.4 or newer the firmware reads the analog values here and there to determine
- open
- closed
- broken / bad voltage range

If you find a solution that fits both ways please feel free to create a pull request.

Please let me know if you need any further information or help.

3d-gussner, 2021-03-18:
@Prutsium As this issue became stale, please consider to close it.
[stale]

closed 2021-08-02 by JohnnyDeer