prusa3d/Prusa-Firmware #3051: [ENHANCEMENT] add default sheets for Satin

mneumark, 2021-03-04:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S+
**MMU Upgrade** - N/A

**Is your enhancement related to a problem? Please describe.**
  There is no steel sheet named after Satin sheets so you have to use Custom sheets
  
**Describe the solution you'd like**
Add a few slots for Satin1, Satin2, et cetera.

DRracer, 2021-03-04:
@mneumark this is a very nice issue. It shows how some users understand the sheet profile slots - no offense, we are glad you reported this. 
Technically: all sheet preset slots are equivalent and differ only by their default names which can be changed by the user (even though typing 7 characters on the knob is a kind of retro game).

mneumark, 2021-03-04:
I know you can rename slots. I already renamed one to Satin. But if you are going to have smooth a textured defaults it's inconsistent to not have them for satin sheets too.

awenelo, 2021-05-31:
@mneumark, release 3.10.0 adds satin sheets (`Satin 1` and `Satin 2`) to the default sheet names list. Please consider closing this issue.

To get the new names, you'll need to reset sheets 5 and 6 (previously `Custom1` and `Custom2`) using `Settings -> HW Setup -> Steel Sheets -> [sheet name] -> Reset`. Note that this will also reset the Z-offsets saved to these sheets.

mneumark, 2021-06-01:
Awesome. Thanks!

PR #3055 (2021-04-22): Add satin sheets
This changes the `Custom1` and `Custom2` sheets to be `Satin 1` and `Satin 2`. Additionally, as there are now only the numbers `1` and `2` at the end of sheets, this replaces the switch that was used to determine sheet numbers, now using modulo to assign the sheet a `1` or `2`. I believe that I have updated the table in `eeprom.h` correctly, if I haven't, let me know. 

To have the sheet names change, the sheet must be reset.

### New:
Index | Name
---|---
0 | Smooth1
1 | Smooth2
2 | Textur1
3 | Textur2
4 | Satin 1
5 | Satin 2
6 | Custom1
7 | Custom2

### Old:
Index | Name
--- [... cut]
closed 2021-06-01 by mneumark