prusa3d/Prusa-Firmware #3053: [FEATURE REQUEST]<Min temp Error Correction For Cold Climates>

DarrenW45, 2021-03-05:
In colder climates the printer (MK3) powers up and goes to error min bed temp. After heating the bed with a hot air gun it clears to hot end min temp, along with the hot end when that is heated.  It's then all good after a restart.

This is a work around that I have to do every day when I turn on the printers and it's becoming somewhat laborious as overnight my building temperature drops well below the 16c threshold for errors, hence the problem every morning.

I know the threshold is there to detect a sensor failure of either hot end or bed but, on start up can this not run a different parameter?? Something along the lines of;

If hot end and bed values are within 2-3c of each other then allow preheat to commence and once each value is over 16c revert to original safety code.

The fact that the printer locks when the min temp error appears eliminates the possibility to select a preheat to get the machine warming slightly.

It's a small niggle but massively common in colder countries and i believe requires sorting as heating a brilliant printer with a heat gun to get it working isn't ideal.

J3r0github, 2021-03-08:
Maybe you could detect the ambient temperature with the ambient thermistor and activate a corrective measure?

DarrenW45, 2021-03-15:
The printer is about to leave my workshop via a window..... Why is a fantastic printer let down by firmware?? The solution is a small change... come on Prusa as I'm sure you can tell I'm getting rather pissed off at having to heat the printers every morning before they get to work.

JamesRHarris, 2021-04-27:
I am seeing this problem with OctoPrint. My printer is in an unheated garage and at night I can hit the low bed temperature warning which cascades into an octoprint disconnection. It seems the low temperature warning is active just when being monitored. Is the fix to only have the warning when actively heating and the heating isn't making progress? Is the point of the monitor to notice an attempt to heat and failing todo so? Or is it some sort of condensation monitor.

DarrenW45, 2021-04-27:
The solution to this is a very simple code change. Having the min temp active only whilst heating would make little difference as you will still have to heat the bed by external means before using the printers own heater.

I've got 5 Mk3's that fail every morning and all get one step closer to being replaced by another brand. I don't like that idea as the Prusa is such a great all round printer but support is close to F... all.

JamesRHarris, 2021-04-27:
> The solution to this is a very simple code change. Having the min temp active only whilst heating would make little difference as you will still have to heat the bed by external means before using [...]

At 5 you have a significant investment to replace. Since the firmware is open source I would suggest you fork the firmware and make the change yourself or pay someone todo it. It would likely be cheaper than replacing 5 printers.

IntegersOfK, 2021-04-30:
I just need an extra ~60 seconds before the timeout, so it has time to heat to above the mintemp threshold. I end up starting the print or pre-heating, waiting for the mintemp error, then I press reset and then for the second time it'll be fine because it has already heated a bit from the first attempt.

magicdroping, 2022-10-24:
I agree with that. The Prusa Mk3s printer itself is a very good machine. But to use hairdryer for the printer and not for the hair isn't very comfortable and also isn't safe. Just would be nice to get like additional mode inthe settings to make printer never go lower like 18oC degrees would be perfect. And everytime it will preheat if it will go lower than 14oC. So we wouldn't get a mintemp error. If you want no heating at all with this mode, just turn off the printer. Will wait for the news.

Prusa-Support, 2022-10-26:
Hello, thank you for the comment, sorry for intervening so late on this thread. Currently, the temperature threshold, on MK3-family, for Mintemp to occur, is 10 °C, 50 °F, for both hotend and heatbed. Was is still problematic? 

Unfortunately, regarding MK2-family, the threshold are higher due to different hardware design of the motherboard (hotend thermistor - temperatures below 22 °C, 71.6 °F, heatbed thermistor - temperatures below 15 °C, 59 °F). 

In case of more feedback, feel free to let me know here, it might be not possible but I'll certainly review this aspect in more detail with devs to see if there might be any possibilities. 

Alessandro Pantaleo
Prusa Research

DarrenW45, 2022-10-26:
Hi Alessandro,

Thanks for your response. The 10c min temp has made it easier to work with but in the U.K. as from November onwards my workshop is below 10c on a morning. Ideally the mon temp would be a float value as long as the hot end and bed temps were equal on power up.
I understand the methodology of the sensor failure detection but if the startup code was an IF hot end && bed temp are = then heat to preset temp.

Regards

Darren Wandless

Patronics, 2022-12-12:
Now that there's the new beta/RC features to improve detection of anomalous temperature circumstances, could this issue be addressed? My printer's also in an unheated garage, with no room to add a heated enclosure for it, and it's really frustrating to need to use a hairdryer with it or hold my hands on the nozzle and heatebed to heat it up before I can start printing. Why can't the firmware have some logic to turn on the heaters briefly if in a mintemp circumstance when the printer has been inactive, and measure whether the temperature increases as it should, and only trigger the error if no heating is detected?

IntegersOfK, 2022-12-12:
I've had it happen in rooms as high as 16°C, which isn't even very cold. I don't think we need to overthink it and we've basically workshopped the answer:

Best Resolution: Watch the increase in temperature as a difference over time, instead of a static threshold after a certain amount of time. Ie. If the temperature has increased by more than 2 or 3 degrees in the past minute and appears to continue increasing steadily, we can assume it's fine and it's just taking a longer time to warm up due to the ambient temperature.

Fine Resolution: Compare the temperature of the bed to the nozzle. If they are both increasing, assume it's fine.

Acceptable Resolution: Increase the time before it times out. That's it, just give it another minute or two to get up to temperature before triggering the failsafe.

Anyway, until this is changed, we all just wait for the error, then restart the printer and it's fine the second time (because it has pre-heated a bit). It's not the end of the world, jus super annoying.

Patronics, 2022-12-13:
Yep, @wavexx  I agree with @IntegersOfK about this, accept that for my case the "Acceptable" option likely wouldn't help much, as I use the printer with octoprint, so it's usually always already on and has long-since triggered the mintemp errors. 

To answer your specific question, it tends to happen for both the hotend and bed mintemp, and the most recent time it happened the ambient temperature was around 10°C, though the environment is certainly sometimes somewhat colder than that, and it likely has happened in somewhat warmer environments as well. 

In this case, the thermistor does accurately represent the temperature, but is just not trusted by the system due to the fact that a failed thermistor tends to give the same sort of range (or that's my understanding at least). To me the solution seems rather obvious and simple, enabling the printer to try to heat up the nozzle and bed (rather than just refusing to due to the error), and then only throwing the error if it doesn't detect the temperature rising at the expected rate, as it would if the thermistor truly has failed or disconnected.

wavexx, 2022-12-13:
> Best Resolution: Watch the increase in temperature as a difference over time, instead of a static threshold after a certain amount of time. Ie. If the temperature has increased by more than 2 or 3 [...]

A problem with the slow increase in temperature is that it can be due to a busted heater or a degraded thermistor (see the "degraded thermistor sim" in https://github.com/prusa3d/Prusa-Firmware/pull/3552). On a cold start the ramp-up time is a lot slower, and because of the thermistor noise at these temps it's a wild guess what's happening (you can have +/-10C, on top of the einsy +/-5C). It's true you can say "ambient temp is low so keep heating" using more sensors, but the thermistor can also be underreporting the whole time so the hotend is maybe actually getting a _lot_ hotter and you can't check this using the einsy or bed thermistor.

IMHO a safer approach is what I wrote before. On a cold start we could keep a low, but safe power level (say a fixed 20% instead of full power) combined with a timeout. On a standard 40W heater cartridge it would settle at <200C so even if the thermistor is shot overheating is not possible. This will require longer to exit from a cold start though, and a timeout is still necessary (if the hotend doesn't recover from X minutes something is also wrong).

wavexx, 2022-12-13:
Another tidbit which would be useful for feedback is interaction. You might start the printer and get into mintemp, but you might also get into mintemp progressively while idling (overnight for example).

Assuming we have a cold-start procedure, what should trigger it? Preheat I guess, if you're in front of the menu. This means mintemp shouldn't immediately lock the printer.
What if the printer is remotely controlled?

IntegersOfK, 2022-12-13:
When I control my printer with OctoPi, I set the temperature to 20°C and once it gets up to that temperature I increase it to the full 215°C to prevent it from triggering the mintemp failsafe / locking me out.

wavexx, 2022-12-14:
@IntegersOfK do you also have a sock?

IntegersOfK, 2022-12-14:
I don't have a sock, I find it hard to get on and stay put.

Evanthx, 2022-12-19:
Well this explains something - I keep my thermostat in my house at 69 or 70F. The hotend error happens if the temperature is below 71.6. So this would be why I can never start my Prusa without hitting the nozzle for a minute with a heat gun!

The average room temperature is between 68 and 74 according to a quick google search. This should probably be set to a bit below the average room temperature, and not set to fail for 2/3 of that average range. As is, the code quite literally is set to fail this check when the printer is at room temperature.

Changing the check to 67 or 65 would resolve a LOT as well as being a simple change.

wavexx, 2023-01-15:
IMHO this is related and needs to be fixed along with #3898

magicdroping, 2023-01-15:
It's getting frustarted over the years that such a simple thing can't be
fixed. It wastes a lot of time if used every day. Even with a new 3.12
firmware nothing changed. I don't know its because of law regulations or
something like that But actually it would be nice that it would let to turn
off mintemp error with somehow until like 4oC.

knmorgan, 2023-01-15:
Why is the new thermal model not able to tell the difference between a cold ambient temperature and a faulty thermistor? The [Prusa blog](https://blog.prusa3d.com/mk3s-3-12-beta-firmware-new-thermal-model-protection-and-blob-detection_71230/) discussing the thermal model discusses some complex thermal anomalies it can detect; detecting a cold environment (either through comparing against other thermistors or from historical data) seems easy by comparison.

wavexx, 2023-01-15:
@knmorgan when the hotend thermistor goes below MINTEMP the temperature readings are increasingly unstable (it's effectively operating out of spec). At that point you cannot distinguish cold temperature from a hardware fault anymore.

Ignoring low temperatures is easy to do, but it's a hack, not a solution. If you want to maintain safety check integrity there are a many places to change where you need to add exceptions to critical logic that _needs_ to work.

knmorgan, 2023-01-16:
When looking at a single data point I understand, but the Prusa blog on the new thermal model implies the printer is able to look at a history of thermistor readings to better understand thermal anomalies. I would assume a steady decline from 15C to 10C over the course of a few hours (like what happens in my garage before MINTEMP is triggered) looks a fair bit different than a faulty thermistor.

magicdroping, 2023-01-16:
Guys, just think simple. If its not possible to lower treshold temp due to
safety concerns. Then add a function to the printer. It will be even more
safe to printer itself.

"Cold environment protection", function if heatbed or hotend temp < 13oC
then heat it up to 18oC. Fairly simple.
And if Prusa team cant do this fairly simple and very useful  I will need
to install an octoprint and try to program and run myself. Somehow

IntegersOfK, 2023-01-16:
@magicdroping What if there's a problem with the heating during "cold environment protection" portion? Just let the printer melt itself to death?

The team is engaging with this situation, and their attempt to work through a solution is fine. Stop being so terse.

And yes, it's easy for you to deal with this yourself, so why don't you go do that? Let's wait for an update on the direction here before complaining further.

magicdroping, 2023-01-16:
Okay, I agree I may have pushed too much. But the goal is about improvement
of the printer . It's about community and we communicate like in a forum.
and that we can communicate and offer what matters users the most is a huge
win.  I don't expect copy cat chinese doing this. I expect a one of the
most advanced teams doing it. It's just my suggestion after working with
prusa printer for over 5 years everyday. Talking about the function, after
reaching 18oc (for example) it will turn off heating and turn on again on
when it goes down until 13oC. And thanks to all Prusa team for keeping the
good printer even better every year by software updates !

DarrenW45, 2023-01-16:
The cold weather/environment function I started this request for is super simple to solve. 
It was aimed at those of us with printers in outbuildings that see temperatures well below 10c daily. On startup, every morning, all printers need a blast with a hot air gun before powering up and starting the days prints, or even just reading stats for the printer. Pain in the arse...

The solution is a simple check of both temperature sensors and act accordingly:
IF hot end and bed are equal, then heat for 20 seconds and check sensor values again. That will be enough for both the hot end and bed to climb above the min temp values whilst the fail safe protocol remains intact and running.

It needs sorting quickly otherwise people will start moving to printers that don't need this mess around before use. Especially with the hype the Bamboo machine are getting.

wavexx, 2023-01-16:
> IF hot end and bed are equal, then heat for 20 seconds and check sensor values again.

To handle all cases discussed here you should start with "IF hot end and bed read UNKNOWN and einsy is LOW", because that's that you can get in a worst case after a restart in a cold environment. Then: "temporarily disable safety checks, apply heat, recheck sensors". After that you might be in:

- one sensor raised, but the other didn't: faulty sensor? faulty heater? still too cold?
- none raised: faulty sensor? faulty heater? still too cold?
- one or both now suddenly too hot: faulty sensor/s?

case 1/2, apply more heat, but for how long? with a faulty thermistor you might read <10C after applying full power for 90s, yet the hotend is already at 300C.

The open-loop approach is good and safe one IMHO, but you need to be able to disable/enable all basic safety checks and re-enable them in safe manner.

Anybody wants to help in this and start writing some code (we're on github after all)? There are many pitfalls in this seemingly simple thing.

knmorgan, 2023-01-16:
I don’t know if anyone in this thread is employed by Prusa but I think the frustration comes from the fact that Prusa has ignored this use-case, printers in cold-ish rooms, for many years. Add to the fact that MINTEMP handling got even worse in 3.12.0 for cold printers and some recent attractive alternatives to the MK3S+ that don’t suffer these shortcomings, and it seems understandable that users are frustrated.

I am happy to throw up a pull request if it would expedite this but I lack the historical context and general knowledge of the codebase that many people here very clearly have.

wavexx, 2023-01-16:
Overall the thermistor is chosen for best characteristics at printing temp, with the downside that the range <10C is next to useless. It sadly conflicts with safety checks at that point, but it's a good compromise. And don't get me wrong, I totally feel the frustration.

> I am happy to throw up a pull request if it would expedite this but I lack the historical context and general knowledge of the codebase that many people here very clearly have.

I contributed the model checker, so I can help in reviewing all the parts that touched that. The rest I have to read and get the context from this thread and the code so I also don't have a magic sphere ;). But it's clear that a downgrade from a safety perspective cannot be accepted in the official FW. The current status is that it refuses to work when it's unsafe or can't know. We are allowed to do this or better, not worse.

MrWFoulkes, 2023-01-16:
The printer ought to boot into a physically safe state (no running the heaters or steppers) because the firmware might have crashed. Some user input (Octoprint included) should still be required before starting the heaters.

Lupo-Grigio, 2023-01-29:
bump... adding a use case. Mk3S in Yakima WA, high desert environment, it is cold and dry for good portions of the year. It's not Yukon canada cold, but I regularly get bit by the thermistors reading temperature below some arbitrary "too low"

JCLemme, 2023-02-01:
Even if we can't do autocorrection in cold environments, can we at least get a manual override? Like, a button that lets me run the heater for a couple of seconds to warm up the hotend?

wmfmontrose, 2023-02-17:
Is this just about the dumbest thing ever; a machine is too cold to turn on a heater to warm it up. Maybe Prusa needs a heater for the heater. Come on Prusa, this is not a difficult problem to fix with a couple of lines of code at startup and done. Why would you even try to justify why this is happening? Just fix it. (The pasted reply was especially irritating.)

gidthup, 2023-02-19:
I sadly see that I am not the only one frustrated with this most annoying problem of the printer (MK2.5s). It is nearly a year round problem, not just winter time.  The additional problem is that the printer stops and moves the head up a few centimeters. Setting the Z-level to 0 so it cannot be lowered manually.  If too many restarts are needed to get the bed heated beyond the min level it can moves above the max hight and it makes a noise similar to drilling in concrete  and wakes the family. 

Can Prusa add a comment to what their thoughts are on this.  Maybe before the 5th of March as it will then be the 2 year anniversary of this issue.

IntegersOfK, 2023-02-19:
Are we all talking about the same thing? I don't really understand where the intensity of the negativity is coming from, because you can literally just wait 30 seconds for the error to pop and then re-initialize the print. If you're using OctoPi, heat it to 25 degrees and then trigger the start of the print, or add startup gcode in your prints to do the same thing (heat to 25 degrees, then continue).

To me this is a completely benign, yes annoying, but tolerable inconvenience.

wmfmontrose, 2023-02-19:
Good morning, Adam.

This is not negativity, it’s frustration with a very simple bug that hasn’t been fixed. I’m not yelling, just stating the facts and the obvious.

I can understand your defensiveness if you can understand the frustration of many of your customers about an easily fixed problem. It’s a moderately long thread in GITHUB and 2.5 owners have the issue so how long has that been? I was copied an email this morning sent to you through GITHUB from another Prusa owner agreeing with my email. (“I sadly see that I am not the only one frustrated with this most annoying problem of the printer”) I’m don’t know what you mean by “you can literally just wait 30 seconds for the error to pop (?) and then re-initialize the print”. It’s 12 degrees C, or about 58 degrees F in my shop, which is hardly cold,  and it’s shown the MINTEMP ERROR for 30 minutes. I’ve never seen it go off without an outside source of heat. I ran a print that paused at night and in the morning, it had the MINTEMP ERROR and wouldn’t re-start. 

I design and build automated machines for a living, so I’ve done a little programming and adding a couple of lines of code is not difficult. I would never ask my customer to modify my firmware code to get my startup routine to work properly. This is basic programming. I suppose the negativity comes from having a machine with a known bug in the firmware for a few years and the known bug is still giving us grief.  It’s too cold to preheat (?) so we must heat up the sensor with another heater before we can turn on the preheater? A bit unnecessary, don’t you think? That’s why we have micro controllers and computers, to automate the processes. Prusa can develop an Auto level routine, but ignoring a sensor at startup is hard? Maybe you could add a feature called PREHEAT?

I don’t know why we are even talking about this…….

 

Adam, If you’re not in programming, please, just forward this to the programing guy.

 

Thanks, Neil

IntegersOfK, 2023-02-19:
Dude chill, I don't even work for Prusa I'm just a user like you. It's open source, go modify the firmware and flash whatever you like. Remove all the safety features entirely for all it matters.

magicdroping, 2023-02-19:
The constructive critics it's what leads to straight and fast problem
identification. After that we can go for a problem solution. It's really a
frustrating thing and can be avoided like Adam said and 12oC is normal
working temperature in some areas. And talking about safety it isn't an
electric vehicle with hundreds of current and voltages so it will not blow
like lithium batteries. Power supply has fuse, Einsy board has fuse it's
like a laptop with 240w power supply. Do we afraid that we turn laptop in
minus celsius temperature I don't think so. I am more afraid making salad
with a knife than leaving a prusa printers printing or sitting turned on.
Hope this get fixed soon as it passed whole winter and not fixed yet, but
got even got crazy beep when reaching 12oC.

wmfmontrose, 2023-02-19:
Sorry Adam, for some reason I thought you were connected to Prusa.

wmfmontrose, 2023-02-19:
I agree. My MK3S is over three years old and I’ve heard MK2’s use the same firmware and have the same problem, so it’s been a lot longer than a year.. 

Not exactly what I expected from Prusa but evidently GITHUB is not the best channel to go through to get to Prusa.

And again, Adam, Sorry for directing that last email in your direction.

 

I’m out.

MrWFoulkes, 2023-02-20:
My interpretation of `temperature.cpp` is that on printers with an ambient temperature sensor, if the ambient temperature is below a threshold then it enables a fifteen second grace period between the heaters turning on and the printer checking for hotend and bed mintemp errors. This is in the function `static void check_min_temp_raw()`.

@IntegersOfK posted:
> I just need an extra ~60 seconds before the timeout, so it has time to heat to above the mintemp threshold. I end up starting the print or pre-heating, waiting for the mintemp error, then I press [...]

@IntegersOfK, are you able to start the heaters for fifteen seconds? Is this the timeout you mention?

For me, on cold days the printer goes straight to mintemp on startup without a chance to turn on the heaters. Reading the code, I struggle to see why, unless the ambient temperature threshold is too high. @wavexx, do you have any insight?

MrWFoulkes, 2023-02-20:
I just remembered that a few weeks ago, it got really cold inside and the printer started fine. This could've been the ambient temperature sensor finally reading low enough to disable the mintemp check until the heaters had run for the grace period (I didn't actually try turning on the heater). In this case the printer displayed four degrees Celsius for the bed and nozzle. I usually have problems when the printer displays below ten degrees, which is what the ambient temperature threshold is meant to be.

If so, it may be that a simple adjustment of the ambient temperature threshold would solve the issue for many of us.

gisforgirard, 2023-03-17:
`TEMP_SENSOR_AMBIENT
TEMP_SENSOR_AMBIENT`

temperature.cpp

you still have to heat it with a heat gun to get it to the mintemp but at least now it shuts the fuck up and stops beeping if its not actively heating lol

gidthup, 2023-03-17:
I have tried  adding the following gcode  at the point just before the actual temperatures are set by the slicer and it works for my situation.

What it does is increase the temperature in small steps. It waits for the small increase and then turns the bed off and on again for the next temperature step.  This off/on step resets the timeout. As long as the printerbed heats up that 1 degree step within the timeout period it will get through the whole sequence.  If the bed is faulty it will still fail in error.
 I started from 12Celcius with 1 degree steps.  After 20 Celcius it switches to 2 degree steps as my printer/environment can handle that. After 30 degrees bed temperature I never have had problem.

; BEGIN ***Fix cold bed temp startup issues **
M104 S180 ; set extruder temp
M140 S12 ; set bed temp
M190 S12 ; wait for bed temp
M140 S0 ; turn off heatbed
M140 S13 ; set bed temp
M190 S13 ; wait for bed temp
M140 S0 ; turn off heatbed
M140 S14 ; set bed temp
M190 S14 ; wait for bed temp
M140 S0 ; turn off heatbed
M140 S15 ; set bed temp
M190 S15 ; wait for bed temp

etc......

M140 S28 ; set bed temp
M190 S28 ; wait for bed temp
M140 S0 ; turn off heatbed
M140 S30 ; set bed temp
M190 S30 ; wait for bed temp
M140 S0 ; turn off heatbed
M140 S32 ; set bed temp
M190 S32 ; wait for bed temp
M140 S0 ; turn off heatbed
M300
M300
; END ***Fix cold bed temp startup issues **

MystaraTheGreat, 2023-03-27:
Argh.   I've just spent the last hour trying to figure out what this error is all about.  I moved my printer to my unheated shed a couple of days ago.  First day it worked fine and now this morning I keep the never-ending beep and errors about min temp.

Like others, I'm using octopi (which connects and then immediately disconnects because of the error).

MystaraTheGreat, 2023-03-27:
Yep.  Just borrowed a heater from my child's room and pointed it at the Prusa.  Once both temperatures were over 15 degrees the error stopped and the print job has started correctly.

Admittedly, the heat bed is heating at only about 0.4 degrees per second, which is perhaps a tad slower than normal.  But with it being a much bigger surface area to heat, that doesn't seem surprising.

Seems odd that the default assumption of "the temperature is low" is enough to conclude that there is an error.

wmfmontrose, 2023-03-27:
I don’t understand what Prusa’s problem is with this. For people whose machine is kept at a steady 76 degrees F, the code is great, but for the rest of the 85% of their customers who don’t live in the tropics, It’s a real pain in the butt. A 3D printer shouldn’t have to come with a hair dryer.

Prusa-Support, 2023-04-20:
> I don’t understand what Prusa’s problem is with this. 

The answer is long and complex.
Find in the page [`ctrl`+`F`] the word _wavexx_ to find some reference highlights posed by the user.
Our associate developer briefly explained only a few of the conflicts and safety concerns.

Regardless of this, our developers and collaborators will look into this. The issue will stay open.

Michele Moramarco
Prusa Research

Ghostbird, 2023-05-01:
Personally I'm running into this issue too in winter. I have a garage with floor heating, but since the gas prices skyrocketed, I've reduced the heat to just frost prevention (7°C). Now when I want to print in winter, I have to warm up the heat bed using my hands, then briefly flick a lighter under the hot end to heat it above 10 seconds.

I feel that a dead man's switch override would be a perfectly safe measure. My proposed process is this:

- Error MINTEMP should only occur when heating.
- When heating is started but the temperature is too low initially, instead of the current "ERROR MINTEMP", prompt: Hold the button to heat beyond MINTEMP".
- Hold down the button, as soon as temperatures are stable above MINTEMP, prompt "release the button to continue normal operation".
- For maximum safety, the normal ERROR MINTEMP can still occur after the button has been held down for a certain time, e.g. after one minute. As far as I know a the Prusa's can't heat from below MINTEMP to dangerous runaway temperatures within a minute.

This is basically makes it possible to use the printer's own heating elements to perform manual pre-heating to a temperature above MINTEMP.

clintonefisher, 2023-10-17:
PRUSA:  How is this STILL not fixed??   It's been YEARS!

Even in the latest 3.13.1 firmware on my MK2.5S, I still get a MINTEMP BED error on initial preheat, then I have to reset the printer, and preheat AGAIN and then everything works fine. 

I LOVE Prusa, but EVERY TIME this happens I get infuriated that SO MANY cold-shop-dwellers have proposed workable solutions and algorithms, including myself, and it seems we are ignored.   I guess the MINTEMP behavior's gotten better, so thanks for that, but it's still not fixed.

Not everybody prints in a heated print farm.  My environment isn't even really that cold - it's 16C and I still get the MINTEMP BED error when starting up cold.

Is it because I'm running the MK2.5S?  Did Prusa forget that the MK2.5S is only a 12v heatbed and it cannot heat as quickly?  (My MK3S works fine at the same room temp).
Can you just lower the expected heating rate for the 12v beds?

Judging by the number of search hits on MINTEMP BED that have nothing to do with bad thermistors or connections, I know I'm not alone.

PLEASE PLEASE FIX THIS for all us cold-shop dwellers!!
Apologies for the heightened emotion, but it sucks that we all still have to deal with this every Fall season through Winter, and into Spring.
Thanks for anything Prusa can do to address this issue.

P.S.  Thanks for the awesome MMU2S firmware fixes!  So much more reliable now.  Looking forward to the next release that address bowden tube length, etc.

Ghostbird, 2023-11-11:
@eefweenink Did you mean M190 or M109?

Anyway, this is a very interesting find. I guess it removes all of the safety the standard mechanism provides. However, it seems that the unsafe way is the only way the printer stays usable in cold seasons.

eefweenink, 2023-11-12:
Goedemorgen Gijs, 

> @eefweenink Did you mean M190 or M109?
It is about M190, the bedtemp. (this is what causes the error here. As soon the room is warmer there is no problem)

> Anyway, this is a very interesting find. I guess it removes all of the safety the standard mechanism provides. However, it seems that the unsafe way is the only way the printer stays usable in cold seasons.
I don't know much about the safety procedures in the firmware, but as I understand all safety is kept:
- The M190 has to wait until the temp asked for is reached. 
If there would be a "major" issue with temperature then the next M190 step will not be reached in time, and the error will be thrown as normal. 

Regards, Eef

Ghostbird, 2023-11-12:
Ah, that's better than I expected in terms of safety.
In your first message it seems you mention M109 a few times where you mean M190 (unless I'm mistaken). If you'd like, you could go to the Github thread, and edit the message to correct that. It would help people who read this conversation later.

eefweenink, 2023-11-12:
> In your first message it seems you mention M109 a few times where you mean M190 (unless I'm mistaken). If you'd like, you could go to the Github thread, and edit the message to correct that. It would help people who read this conversation later.

Thanks, I have changed it. 👍

gisforgirard, 2023-11-20:
https://github.com/gisforgirard/Prusa-Firmware/tree/without-MINTEMP

i see no purpose that mintemp serves whatsoever so here it is removed if anyone wants it

Prusa-Support, 2023-11-24:
Printing in very cold ambients without a thermally controlled chamber is not recommended but theoretically possible.
Some of the concerns are explained above in the thread - https://github.com/prusa3d/Prusa-Firmware/issues/3053#issuecomment-1516735486 - however, our developers are making considerations about possible safe ways to go around the obstacle.

The MINTEMP error would occasionally be triggered if a thermistor is disconnected or damaged.
Please note that by removing the MINTEMP safety features you are drastically lowering the protection barriers and exposing you and your printer to serious hazards in case of loosened or damaged thermistor. The Thermal Model protection is a big hand here but safety shouldn't be taken lightly anyway.
 
Michele Moramarco
Prusa Research

gisforgirard, 2023-11-24:
> Printing in very cold ambients without a thermally controlled chamber is not recommended but theoretically possible. Some of the concerns are explained above in the thread - [#3053 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/3053#issuecomment-1516735486) - however, our developers are making considerations about possible safe ways to go around the obstacle.
> 
> The MINTEMP error would occasionally be triggered if a thermistor is disconnected or damaged. Please note that by removing the MINTEMP safety features you are drastically lowering the protection barriers and exposing you and your printer to serious hazards in case of loosened or damaged thermistor. The Thermal Model protection is a big hand here but safety shouldn't be taken lightly anyway.
> 
> Michele Moramarco Prusa Research

while you are probably correct, i have considered your points and already addressed them in my temperature.cpp changes, see lines 1485-1487: https://github.com/gisforgirard/Prusa-Firmware/blob/b0ff16243ed96a26957a31270cf3d35510736b30/Firmware/temperature.cpp#L1485

furthermore, i looked at the code and i am confident the thermal model will trigger so personally i am not worried about it and consider MINTEMP obsolete with the addition of the thermal model and simply a relic that causes the printer to in many other circumstances perform unacceptably worse.

3d-gussner, 2023-11-25:
> i have considered your points and already addressed

Your commit may sound funny, but it is one thing if you modify your custom firmware and disable few safety features at your own risk and totally another when Prusa would do that.

gisforgirard, 2023-11-25:
> > i have considered your points and already addressed
> 
> Your commit may sound funny, but it is one thing if you modify your custom firmware and disable few safety features at your own risk and totally another when Prusa would do that.

You're right. I'm just a frustrated user here posting my solution for similarly frustrated users to consider as a potentially available option, and based on the history of this issue I have zero expectation that it will ever be addressed by Prusa.

Let me point out, though, that this feature is extremely poorly implemented, and it provides *ZERO* safety by triggering when ALL HEATERS ARE TURNED OFF AND THE PRINTER IS COMPLETELY IDLE. I understand why this would be misunderstood by those who aren't attempting to operate their printer in low ambient temperatures, but for those of us who are, this is literally an issue that turns an otherwise fantastic device into a device that I want to drop kick into oblivion and go back to using my ender 3, which ironically is a far worse experience in every other aspect but has no such MINTEMP issue. My solution is FAR from optimal, but I assure you that it has made me happy to have a functioning printer again without this feature implemented. Hopefully one of these days someone who is more competent than I can come up with an acceptable solution, heck, it could even be you... actually, isn't that the point of this whole open source thing to begin with? Thanks for your input though.

[105 more replies]

closed 2025-03-15 by Prusa-Support