prusa3d/Prusa-Firmware #3059: [FEATURE REQUEST] Maintenance menu for the I3 mk3s

J3r0github, 2021-03-08:
I would like the printer to have a maintenance menu where you could for example select the printer to move all axises to apply lubricant to the smooth rods. Currently you have to move the axis from the menu, this is easy enough but this would be a better (and slightly faster solution) it also could have some instructions as to how to apply lubricant. Also from the menu you could toggle a message like #2377  which would remember you to lubricate the rods, clean the sheet and that sort of stuff every _XX_ printing hours you could also toggle the printer to automatically run selftest every _XX_ printing hours to test that all of the components are still working. The menu could also have some other functions. It should be in the support menu maybe? I would code this myself but i have no idea how to code prusa firmware. Sorry for the indetailed post, this is my first github issue.

DRracer, 2021-03-18:
this is planned...

fe60, 2023-10-13:
The issue #2377 requesting a maintenance reminder has been closed by a bot and a [reply from PrusaSupport](https://github.com/prusa3d/Prusa-Firmware/issues/3483#issuecomment-1331140248) in #3483 says, the maintenance feature will be added to 

> the new generation boards (like the Buddy 32-bit board)

and 

> Later on, Prusa Connect should support similar features

3d-gussner, 2023-10-13:
We are working on that probably for fw 3.14.x

girak49, 2023-12-09:
I would like to add more reasoning to why such a maintenance reminder feature is even more valuable/necessary today. 
The MK3 & 4 are marketed as reliable work horses. On the [MK4 homepage](https://www.prusa3d.com/product/original-prusa-mk4-3d-printer-8/), the first sentence that Prusa uses to explain it is "The MK4 is the latest edition of our flagship 3D printing workhorse." Regular maintenance is a must if reliability is expected. And now that input shaping is available on all new printers, this puts more stress on the printer. A quote from [Input Shaper Prusa Knowledge Base:](https://help.prusa3d.com/article/input-shaper-mk4-xl-mini-_451816) "We strongly suggest performing basic maintenance on the MK4 before you start to use the Input shaper to achieve the best possible print performance." Our cars remind us when to change the oil, I think our printers should too! :)
A simple reminder every 200 print hours to perform the recommended maintenance would be greatly appreciated. Thank you!