prusa3d/Prusa-Firmware #3060: [FEATURE REQUEST] Make the printer go to stealth when not printing

J3r0github, 2021-03-08:
Simply, when the printer finishes printing, if the user has set a setting like _Silent Idling_ when the printer finishehs printing. 
Dont get me wrong, its silent as it is but it doesnt cost anything to put it in stealth mode when its not printing, and if you can do that it will be a tiny bit quieter.

DRracer, 2021-03-18:
@J3r0github I'm not sure if I understood correctly - the MK3/S/+ already switches the Z-motors into silent mode to lower even the noise coming from the PSU automatically when not printing. 

What else do you expect the printer does? Fans are off, motors are not moving, heaters are off... the only thing running at idle is the motherboard and that is completely silent.

brettvitaz, 2021-06-28:
I have noticed this as well. When not in silent mode and not printing, the stepper motors whine. This is especially noticeable on the MMU2S for some reason.

wavexx, 2021-06-28:
I think this might be due to "stepper hold". If you go to "Settings -> Disable steppers" do you notice a reduction in noise?

JohnnyDeer, 2021-08-19:
I'm closing this issue because there are no responses.

closed 2021-08-19 by JohnnyDeer