prusa3d/Prusa-Firmware #3063: [BUG] extrusion stops for a while during print

hajkys, 2021-03-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- 3.9.3-3556

**MMU Upgrade** - none

**SD card**

**Describe the bug**
Print starts normally after bed calibration. skirt (outline) starts printing and about after 6cm extrusion stops for several seconds, but print head moves on like printing. Then the first layer sequence starts, still no extrusion. After a while, extrusion is again started. Few seconds later, still in the first layer, extrusion stops for few seconds once more. Then the rest of the print continues normally.

Result is that there is a gap in the first layer and skirt is not complete.

[image]

More info on print:
Polypropylene, nozzle 1mm, nozzle temp 180 (same result with 190), bed temp 60
  
**To Reproduce**
It is deterministic problem - I have sample gcode which shows this issue exactly at the same point in the beginning of the print, again and again. I have checked the gcode at that suspicious places and the extrusion instruction E is there, seems to be correct.
  
**Expected behavior**
Extrusion should be continuous.
  
**G-code**

[hledacekspodekpokus2.zip](https://github.com/prusa3d/Prusa-Firmware/files/6111384/hledacekspodekpokus2.zip)

**Video**

[video](http://www.uschovna.cz/zasilka/IZ86P8JX9V543C7N-984)

hajkys, 2021-03-09:
Just downgraded firmware to 3.9.1-3518 and behaviour is almost the same. The first pause is exactly at the same point and approx of the same length, but the second one does not happen at all.

DRracer, 2021-03-18:
@hajkys looking at your photo - the primary issue is probably the fact, that your Live-Z is too low. The filament is squished into the print sheet and your nozzle probably clogs.

JohnnyDeer, 2021-03-23:
@hajkys is it the same when you place the model in the middle of the bed? Or if you are printing another model that covers most of the bed?
If not, it is due to the incorrect Z layer height, not the FW issue.

JohnnyDeer, 2021-03-30:
So filament is barely attached to the bed and extrusion still stops, right? Could you please upload the video once more, is not available on zásilkovna now, thank you.

hajkys, 2021-03-31:
Right. Video uploaded again. It is very deterministic, I ran the gcode several times, and it is always exactly the same. Maybe it is slicer issue.
Also I did not experience this when using normal 0.4 nozzle, only with 1mm one. Printer also does not allow 1mm nozzle selection, only 0.6mm is closest, but question is, whether this can have any influence on gcode interpretation..

DRracer, 2021-04-08:
@hajkys Now we are getting somewhere.
- Your printer uses a 1mm nozzle. This should not be a problem, the nozzle size selection in the FW is just for checking the G-code (there is no computation in the FW that reflects the nozzle size)
- G-code preview in PrusaSlicer didn't show anything wrong ... and looking at the G-code directly also doesn't show anything unusual
- Your extruder really doesn't move at some spots, that is the key problem, I'm still unsure why this happens.
- @JohnnyDeer reported to me, that he is able to reproduce your issue with your G-code as well, so it really looks like a FW issue. Please stay tuned, we are looking into it.

JohnnyDeer, 2021-04-08:
@hajkys We figured out what's wrong here. In your Gcode is nozzle temperature set to 180°C. When on begin of the print the print fan turns on, the nozzle temperature drops below 175°C and extruding is stopped by safety feature which should save extruder in case that heater or thermistor is not working right. Please try to set your nozzle temperature bit higher and print should work right. Let us know if it helps, thank you.

hajkys, 2021-04-16:
Well - that was it...
Now I raised the 1st layer temp from 180 to 200 and it prints without any issue. I have observed the nozzle temp drop down by 6-7°C when fan is on, absolutely correct...

But I have to print PP with 1st layer @180 otherwise it welds with the PP tape so that it cannot be separated safely.

I will try adding an instruction to run the fan from the very beginning and then wait for 180 to stabilize.

I was just wondering - would it be possible in general to pause the whole gcode - to stop all axes - when temp drops below safe level - and resume once it gets up again? So that printer would simply wait on the spot for the nozzle temp to continue..

Thanks once more everybody for help!

DRracer, 2021-04-19:
@hajkys you can lower the cold-extrusion prevention temperature by using M302, but you have to be absolutely sure that your filament handles the lower temperature correctly.
https://reprap.org/wiki/G-code#M302:_Allow_cold_extrudes

Waiting for a target temperature is done with M109, but that doesn't take PID regulation into account. So in your case I'd try the M302 carefully, for example M302 S170.

Anyway, as this is not a FW issue, would you mind closing it?

hajkys, 2021-04-19:
Thanks for this advice.
I do not mind closing this issue then.

How about the idea of stopping all axes in case of cold extrusion? Would it make sense? As a feature request, not bug of course.

DRracer, 2021-04-19:
@hajkys stopping all axes in case of cold extrusion will bring a wide range of edge cases. It won't solve your problem correctly though, as stopping the extruder at a place and wait for heat up will surely create blobs on the print.

Anyway, thanks for your confirmation, closing this issue

closed 2021-04-19 by DRracer

hajkys, 2021-04-19:
Thanks a lot for your support!