prusa3d/Prusa-Firmware #3069: [BUG] Heating phase does not wait for cooldown when bed is too hot

gtozzi, 2021-03-17:
**Printer type** - MK3S
**Printer firmware version**-  3.9.3

**MMU Upgrade** - no

**SD card or USB/Octoprint** - USB

**Describe the bug**
When the bed is already above the desired temperature during the heating phase, the print starts immediately instead of waiting for the bed to cool down.
  
**To Reproduce**
- Start a high temperature print (like PETG), wait for bed to reach 90°C
- Realize that you've chosen the wrong settings for your filament, abort the print before any filament is extruded
- Start the print again with a target bed temperature of 60°C
- The new print will start immediately with the bed still at 80+/60°C
  
**Expected behavior**
I would expect the printer to wait the bed too cool down to the desired temperature before starting the print.
Same may apply to the extruder temp: didn't test it, but the extruder has less thermal inertia so the problem is less common in practice.
  
**G-code**
Not relevant

NStarkebaum, 2021-03-18:
@gtozzi, 
The default start G-Code in PursaSlicer for the MK3S will contain this line:
`M190 **S**[first_layer_bed_temperature] ; wait for bed temp`
The S in the M190 command means "wait until the bed is at or above this temperature". M190 also has an R command which means "wait until the bed reaches this temperature, even if cooling".
If you replace that line in the start g-code with this, it should fix your issue:
`M190 R[first_layer_bed_temperature] ; wait for bed temp`

The M190 command's documentation is located at: [https://marlinfw.org/docs/gcode/M190.html](https://marlinfw.org/docs/gcode/M190.html), this site is a great resource for learning new tricks. For example, there is also an R available for M109 which sets your hotend temp the same way: https://marlinfw.org/docs/gcode/M109.html. https://www.reprap.org/wiki/G-code is also a great resource for G-Code information. 

I apologize if you already knew this and are reporting an issue with M190 R not working as intended.

3d-gussner, 2021-03-18:
@NStarkebaum Thanks for the fast and correct answer! Love to see when the community is helping each other.

@gtozzi Please consider to close this issue.

gtozzi, 2021-03-18:
Sorry i'm a newbie, i believed it was the firmware, not the G code, controlling the initial setup.

So, this is not a firmware issue, thanks @NStarkebaum for the explanation!

closed 2021-03-18 by gtozzi