prusa3d/Prusa-Firmware #3072: [FEATURE REQUEST] Filament Pull Through when filament runs out

TurboCooler, 2021-03-18:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK3S, MK3]
**MMU Upgrade** - [MMU2S, MMU2]

**Is your feature request related to a problem? Please describe.**
  This has been a persistent problem for me.  When you are using a filament spool from any manufacture and you are at the end of the spool, while the Prusa detects that the filament has run out, the problem is that the filament is NOT always properly ejected.  Often the filament during ejection, because the end is bent or somehow deformed, will jam inside the extruder.  I attempt to dislodge the jammed filament, but after the jam is cleared and the new filament is replaced, the Prusa does not properly return to where is last started to print and thus causing my print to fail.  When the filament does properly eject (a rare occasion) everything works as designed and printing returns to normal.
  
**Describe the solution you'd like**
A better solution I believe would be for the following process to occur.  When a filament run out has been detected, home the extruder as it does today.  Run a routing to expel the remaining filament.  What little is otherwise ejected is to small to be useful anyway and will be throw away, better to extrude all remaining filament in the extruder than to risk a failed print or jammed filament.  This is done in many industrial machines for the same reason, better to purge the filament than reverse it back in and cause a jam or failure.  Only enough filament needs to be extruded to allow the new filament to catch on the bondtech gears and then the new filament can be extruded as normal.

I have had several 14hr prints fail on me because of this issue.  There is no real reason to eject the small piece of remaining filament and it would be best for all if we simply ejected the small amount that remains inside the extruder when filament has run out.   This would save everyone from having a failed print.  In a print farm situation, this is the best solution because the extruder will home, extrude the remaining filament and then it remains in a wait state for a technician to simply add new filament and the chance of a print being lost is greatly reduced.

DRracer, 2021-03-19:
@TurboCooler at least one part of your proposal makes sense - currently the FW expects you do not move the X or Y axis at all while trying to remove stuck filament. So doing a rehome after an "unsuccessful" unload makes sense.
Otherwise we cannot push the remaining piece of filament into the nozzle, because most probably you won't be able to insert any new (look at the gap between the extruder gears and the PTFE tube into the heatblock).

TurboCooler, 2021-03-19:
@DRracer - I will be happy to have anything that prevents a failed print after clearing a failed filament ejection from a filament runout condition.

islander2013, 2021-04-24:
Totally agree...I can't use filament that's almost empty because of the same issue...every brand except prusament jams on the little bend at the end of the spool... I have ruined hours of prints because of this. Can you implement it so as soon as the filament sensor detects a run out, it backs it up a little so the bend doesn't snag? Thank you.

rca3d, 2022-02-20:
I have the same issue, was there a fix on it?

yoyo42, 2022-03-18:
Clearing an unsuccessful unload will nearly always lead to the hotend being manually moved (accidentally by pushing it or deliberately to get access) so the rehoming in X and Y before restarting the print is an excellent idea. The Z axis shouldn't be as susceptible to forced movement and should also be at a safe and known height due to the attempted filament change so just homing X and Y should work.

lifewithmy3Dprinter, 2023-12-06:
I have the same issue. I support the suggested fixes.