prusa3d/Prusa-Firmware #3074: [QUESTION]Pause/Resume functions improved in 3.10 release open a possibile resuming from octoprint?

gorobey, 2021-03-19:


gorobey, 2021-03-19:
I intend on power loose:
I have not tried, but at moment, if i pause a print and power off, i can car resume work, it is correct?
With new feature is possible "automate it"?!

3d-gussner, 2021-03-20:
At this moment please add
`M601` to After print job is paused: in OctoPrint
to get Pause/Resume from SD and Octoprint working

DRracer, 2021-03-25:
@gorobey no, powering off the printer does not keep the paused print at all. This is NOT a pause/resume function.

The only similar situation is called the Power Panic which in fact does what you want, but the Octoprint cannot keep up AFAIK.

zeroflow, 2021-03-27:
> At this moment please add
> `M601` to After print job is paused: in OctoPrint
> to get Pause/Resume from SD and Octoprint working

For me, this setting works when Pausing with the pause button, but not if I put M601 into the GCode.
This setting + M601 in gcode triggers repeated pauses, which mess everything up.

My understanding is:

GCode M610 -> Printer pauses -> Config M601 is queued in Octoprint -> Printer resumes (M602) -> Queued M601 is sent to printer -> Octoprint keeps sending gcode, printer continues moves, but hotend is turned off

closed 2023-11-08 by github-actions[bot]