prusa3d/Prusa-Firmware #3077: [BUG] Printer fails to print certain models after firmware update to 3.10.0-RC1

lhecker, 2021-03-21:
**Printer type** - MK3S+
**Printer firmware version**-  3.10.0-RC1
**SD card or USB/Octoprint** - stock SD card
**PrusaSlicer** - 2.3.0

**Describe the bug**
My MK3S+ fails to print the popular [Cam Clip](https://www.prusaprinters.org/prints/27216-bag-clip-print-in-place-cam) model using the 3.10.0-RC1 firmware. It works finder under 3.9.3.
The initial nozzle flush completes just fine. Afterwards it just prints the first 2-3mm of the skirt and stops entirely. It doesn't pause, but rather aborts the print at that point (there's no pause/stop print menu point anymore). The SD card menu point reads "no SD card", but if you click on it, it'll open just fine.
This issue didn't happen with any of the models I've printed in the days before on 3.10.0.
  
**To Reproduce**
Flash the 3.10.0-RC1 firmware on a MK3S+ and try to print the attached .gcode file using Prusament PLA.
As mentioned above, the print will abort after printing the first 2-3mm of the skirt.
  
**Expected behavior**
The print shouldn't abort.
  
**G-code**
[cam_clip.zip](https://github.com/prusa3d/Prusa-Firmware/files/6177520/cam_clip.zip)

P.S.: I hope this tickets isn't a duplicate. The other, similar issues seemed to be unrelated to this.

Barnabasek, 2021-03-22:
I am able to reproduce this bug with your g-code from that zip file. But not with some of my own g-codes.
- run the print
- stop the printing via the menu during the first layer
- run the print again right away
- it then stops after the purge line showing no SD card in the menu (but you can access the files) and the printer just keeps heating while showing the main screen. Only reset helps.

DRracer, 2021-03-25:
@lhecker we managed to reproduce the issue with your G-code, interestingly the behavior of the printer varies.

Hauzman, 2021-03-25:
I had similar issues with one of my gcode making the brim is strange way and it move the extruder almost in home position and the mesh leveling wash on the lcd calibrating z 49 point and the up down  movement was super slow almous i think the printer was stoped but after a couples of tryouts and reslice the model all was ok again
Here is a video 
https://www.dropbox.com/s/x1jwtozyljqhznr/20210325_082711.mp4?dl=0

Pyroschnueffler, 2021-04-01:
Hello i have the problem my printer starts bevor heating the Bed and the Hotend.
With Firmware 3.9.3 no Problems. That Files that i printing are finished GCodes from a Model.
pls help me i print a RC Plane.

DRracer, 2021-04-17:
Could you please check your test cases with the most recent FW 3.10-RC2 and report back if your issues have been fixed?

Pyroschnueffler, 2021-04-18:
Hello the prints starts, problem what i see, the preheat doesnt work correctly, he prints bevor reaching it. I give you the Code that i printed, its a finished GCode Cura Profile from the Plane. So can you check what wrong is it. Sry i am not a pro in 3D Printing. 

https://www.dropbox.com/s/ekhlhv58snnnp2c/Fus3.gcode?dl=0

DRracer, 2021-04-19:
@Pyroschnueffler interesting test case, looks like M105 is preventing the machine from waiting for the temperature.
```
M140 S60
M105
M190 S60
M104 S220
M105
M109 S220
```
When I comment the M105 lines out, the machine waits for the temperatures as it should.
I can confirm FW 3.9.3 waits correctly. 
It doesn't seem to be related to the original issue posted by @lhecker

PR #3091 (2021-03-30): SDFile - fix errorneous offset computation
... my fault, I was originally too optimistic about the overflow. Instead, it was able to put a '\n' into the block buffer breaking the G-code.

Fixes #3077

PFW-1233
commit cdcc06f3 (2021-03-30): SDFile - fix errorneous offset computation

... my fault, I was originally too optimistic about the overflow

Fixes #3077

PFW-1233
closed 2021-04-19 by DRracer