prusa3d/Prusa-Firmware #3080: [FEATURE REQUEST] PIN Security access to printer on power up

Spotburst, 2021-03-22:
I'm managing a roomful of Prusa i3 Mk3S+ printers in a busy lab and it would be useful if I could restrict usage to trained personnel.
A simple way of enabling a lock PIN would be useful upon printer PWR UP.

Hooch180, 2021-03-30:
I think that for lab. Best option would be to use something like key operated power switch. [Like this](https://www.google.com/search?q=key+operated+power+switch&tbm=isch&ved=2ahUKEwjzwt2GwdfvAhUIyyoKHfJOBfIQ2-cCegQIABAA&oq=key+operated+power+switch&gs_lcp=CgNpbWcQAzoECCMQJzoECAAQQzoCCAA6BggAEAgQHjoECAAQGFDcbFiujAFgx40BaAFwAHgAgAFfiAGqD5IBAjI2mAEAoAEBqgELZ3dzLXdpei1pbWfAAQE&sclient=img&ei=JdViYLOcDoiWqwHynZWQDw&bih=977&biw=1827)

You can either mount it on the AC cable or somewhere on the printer and just cat 24V from the supply to the power board.

3d-gussner, 2021-04-01:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2367
@Spotburst The existing resources are limited and not sure if Prusa will sacrifice some of these for this request instead using it for other more requested features.
Can describe exactly the process you are looking for?
Something like:
1. Power on
2. LCD locked
3. Press knob
4. ...
5. Timeout and lock
6. Back to step 2

How should a "reset PIN" work?

Spotburst, 2021-04-01:
My initial thoughts are that the sequence you've indicated would be good.
Enabling or Resetting a PIN could work via PrusaSlicer (a basic reset or remove PIN feature would suffice)

Anything more complex could cause a 'locked out' printer

The other suggestion (of physical locks) are either easily bypassed (pwr leads swapped) or key op switches require fitting 'into' the casing - best avoided.

Thanks for taking the time to even consider it a possibility (I appreciate there's probably more important mods & requests).

closed 2023-11-08 by github-actions[bot]