prusa3d/Prusa-Firmware #3083: M112 Emergency Stop when printing Multi-material print from Octoprint FW 3.9.3

Phantom1911A, 2021-03-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.3
**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**SD card or USB/Octoprint**
  Octoprint

**Describe the bug**
 When printing a multi-material print, printer will halt with a M112 Emergency Stop displayed on screen.  Octoprint will report MINTEMP Bed triggered.  After reset the issue happens at the EXACT same point in the print.  Will print successfully if printed from SD card
  
**To Reproduce**

  
**Expected behavior**
 
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
[Among Us Bookmark New.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/6186583/Among.Us.Bookmark.New.gcode.txt)
(just remove the .txt from the file name)

[octoprint.log](https://github.com/prusa3d/Prusa-Firmware/files/6186645/octoprint.log)
[Octoprint Terminal.txt](https://github.com/prusa3d/Prusa-Firmware/files/6186669/Octoprint.Terminal.txt)

**Video**
  Please attach a video. It usually helps to solve the problem.

https://user-images.githubusercontent.com/81205857/112087582-9c49ba00-8b64-11eb-9c9a-63d9abf96b8a.mp4
[image]

JohnnyDeer, 2021-03-23:
Hi @Phantom1911A, could you please keep eye on printer LCD in the exact point where the problem occurs if the bed temperature drop down? (even if just for second)

Phantom1911A, 2021-03-23:
> Hi @Phantom1911A, could you please keep eye on printer LCD in the exact point where the problem occurs if the bed temperature drop down? (even if just for second)

It’s in the video. No temp drop.

awenelo, 2021-03-23:
> It’s in the video. No temp drop.

@Phantom1911A, it appears that the temperature dropped on the bed, resulting in OctoPrint calling M112.
If you look at the video slowly, you see the 70 starts to change before the LCD updates to display the M112 message:
[image]
Then, when the LCD starts updating to show the M112 message, the 70 has already dropped to 0: (The 7 is a lot more faded than everything else)
[image]
The temperature dropping is also shown in the log file in the last 2 temperature reportings:
`| Recv: T:215.5 /215.0 `**`B:69.9 /70.0`**` T0:215.5 /215.0 @:33 B@:53 P:48.1 A:39.2`
...
`| Recv: T:214.9 /215.0 `**`B:63.0 /70.0`**` T0:214.9 /215.0 @:46 B@:127 P:48.1 A:39.2`

The M112 is automatically called by OctoPrint when it disconnects because of a printer error:
`2021-03-23 02:29:35,294 - octoprint.util.comm - INFO - Force-sending M112 to the printer`
 If you disable calling M112 on a disconnect due to an error (on OctoPrint), you should see the printer start displaying MINTEMP BED on the LCD.

Phantom1911A, 2021-03-23:
What could be causing it?  It only happens when printing from Octoprint, only on a multi-material print, and always at the exact same point.  If printed from the SD card, it prints successfully every time.  That doesn't sound like a hardware issue.

Phantom1911A, 2021-03-23:
> @Phantom1911A, it appears that the temperature dropped on the bed, resulting in OctoPrint calling M112.
> If you look at the video slowly, you see the 70 starts to change before the LCD updates to [...]

What could be causing it? It only happens when printing from Octoprint, only on a multi-material print, and always at the exact same point. If printed from the SD card, it prints successfully every time. That doesn't sound like a hardware issue.

awenelo, 2021-03-23:
If you try disabling the setting to call M112 in OctoPrint, what happens to the bed temperature at the point that OctoPrint would call M112? (Does the temperature jump right back up, or does it stay at 0?)

It could be that a very specific placement of the bed leads to the bed's wires momentarily disconnecting.

Phantom1911A, 2021-03-23:
> If you try disabling the setting to call M112 in OctoPrint, what happens to the bed temperature at the point that OctoPrint would call M112? (Does the temperature jump right back up, or does it stay [...]

I'll give it a try when I get home tonight and see what happens.

JohnnyDeer, 2021-03-23:
Sorry I missed the video. @awenelo thank you for screens, yes the bed temp is dropped right before M112. Definitely caused by broken bed thermistor. @Phantom1911A please contact our Tech Support for help (chat in right down corner https://shop.prusa3d.com/en/ ).

JohnnyDeer, 2021-07-19:
@Phantom1911A any update on this issue? Do you have new thermistor?

Phantom1911A, 2021-07-27:
> @Phantom1911A any update on this issue? Do you have new thermistor?

I don't have a spare thermistor at the moment.  But as I said in my original post, it ONLY happens when I print a multi-material/colour print.  I've only been printing in one colour for the last while, so the error hasn't come back.

sarf2k4, 2021-12-18:
I have this issues earlier

prusa mk3s+mmu2
Firmware: 3.10.1-4697

hardware: Raspberry Pi 3 Model B Rev 1.2
octoprint: 1.7.2
octopi: 0.17.0

I wasn't sure if this was the octoprint issues or the printer's issue. My M112 were called usually around heating up the bed which after selecting the desired filament/toolhead to use for the current print. I often do single material/filament printing but I get M112 several times in the past as well.

Judging from the current firmware's behavior about minbed temp or something, it is as if the prusa firmware were made to be super sensitive about this. Somehow i kind of think its best to have like a delay like 1 or 3 seconds thermal runway check before really throwing M112

For my case, I often just leave the printer right after selecting the desired toolhead to print. The printer always right besides me but I wouldn't notice if there were M112 issued cause I already silenced all the buzzer noise.

closed 2023-11-08 by github-actions[bot]