prusa3d/Prusa-Firmware #3085: [ENHANCEMENT] Manual temperature entry should have a non-zero initial value

domverity, 2021-03-26:
MK3S - When there is no heating explicitly set and you go to Settings -> Temperature -> Bed | Nozzle and click the wheel the enter a temp, the initial value is zero, but I would suggest that the majority use-case is setting bed to >=60 and nozzle to >=200, which causes excess wear-and-tear on the rotary dial spinning it through (mostly) useless numbers.

Panayiotis-git, 2021-03-26:
Nice proposal. Relates to the https://github.com/prusa3d/Prusa-Firmware/issues/2733

gudnimg, 2021-06-20:
I don't see why this would be needed. If you start a print then the printer will raise the temperature as set in the Prusa Slicer. I think it makes more sense to keep it as is.

Only use case I can see at the moment is if you want to heat up the nozzle to clean it/change it. Then its probably quicker to use the Preheat menu. (I think we could add an option to only preheat the Nozzle save energy 🤔 )

Say we made this change... what if I start changing the temperature but then want to cancel? I would need to scroll all the way back to zero or turn off the printer.

domverity, 2021-06-20:
@gudnimg as you say, the use-case here is non-printing purposes, like nozzle change. Nozzle-only preheat (and in fact bed-only preheat, for other reasons) would be a useful alternatives, as would jumping in increments of say 5C instead of 1C.

mix579, 2021-12-23:
I *love* this idea. Every time I change a nozzle—and I do it a lot—I'm irritated that I have to spin the wheel like crazy to go from 0 to 285 degrees. Starting with 60 or even 80 would make a lot more sense.

PR #3494 (2023-08-02): Jump to PLA temps in LCD temperature settings when starting from zero
When turning on one heater manually through the Settings -> Temperature menu, it's painfully slow to turn encoder to reach _any_ useful temperature (usually, a minimum of 180C is required).

This PR allows to set an initial "jump" value when starting from zero. It's used in the Settings -> Temperature menus for both the Nozzle and Bed, setting the jump point to 215/60C respectively.

That is, when the nozzle temp is zero, and the user is trying to increase the value, immediately set the value to 215 for further editing. The  [... cut]
closed 2023-08-02 by 3d-gussner